prusa3d/PrusaSlicer #407: Shortcuts: Movement in 3D Space

m-Th, 2017-07-08:
### Version
Slic3r 1.35.5

### Behavior
In View menu we have several commands without shortcuts. Here are some, modeled from Blender:

Iso - Num 0
Top - Num 7
Bottom - Num 9
Front - Num 1
Rear - Num 4
Left - Num 2
Right - Num 3

_Is this a new feature request?_
yes

bubnikv, 2017-07-12:
Do you have a link to the Blender keyboard short cut reference? I found the following
https://en.wikibooks.org/wiki/Blender_3D:_HotKeys/3D_View/Object_Mode
but it seems to differ from what you have collected.

Do you know if any other application follow the Blender keyboard scheme?

m-Th, 2017-07-12:
Yes, it differs. Blender's shortcuts which are different have Ctrl in them. But do as you wish.

No, I don't know but it seems to be the most popular program for 3D printing. A rather accurate source can be found here: https://i.materialise.com/blog/top-25-most-popular-3d-modeling-design-software-for-3d-printing/

There are programs which even doesn't have such shortcuts.

An interesting solution is SolidWorks which have Alt+Arrow keys in order to rotate the model parallel with the viewing plane but it isn't what we need best.

bubnikv, 2017-07-21:
Implemented with https://github.com/prusa3d/Slic3r/commit/3b51f644110978f69edaf95b50a8796e992aab23
the key assignment is linear with the order of menu items, so it is not compatible with Blender, but it is straightforward.

Also changed the accelerators for adding / removing duplicates from
Ctrl++/- to plain +/-, from Ctrl-Del to plain Del,
and added an 's' key accelerator for uniform scaling.

closed 2017-07-21 by bubnikv