prusa3d/PrusaSlicer #408: Cut Here

m-Th, 2017-07-08:
### Version
Slic3r Prusa Edition
Version: 1.35.5-prusa3d-win64
Build: Slic3r-1.35.5-prusa3d-win64-full-201706151818
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 6.1.7601] 

### Behavior
In Cut window we do not have a precise view of the layers. 

...however we can fake it with ease. We need two buttons: One in Layers tab and a brother of it in Preview tab.

Add a **Cut Here** button in **Layers tab** in the place marked with red.
[image]

Which will simply load the Cut window (if there is an active object) with the index value of the current layer already loaded as the initial cut value.

For the **Preview tab** I think that we need only one button for the upper layer because we see only this layer from above.

_Is this a new feature request?_
yes

Sebastianv650, 2017-07-09:
A good idea, but it has one drawback: In the layers tab, you will see only the layer resolution given by the layer height setting. While this might be OK for your needs, it's not optimal for every case. For example, when I'm using the cut feature it's nearly every time due to two cases:
- Get a tiny portion of a part for a test print. In this case, it's not important if the cutting plane is +- some fractions of a mm.
- Cut the lower part of an object, because it was not designed with having 3D printing in mind (for example an animal with round feets, but we need a flat lower surface for printing). In this cases, I wan't to cut as little as possible, therefore I'm moving the cuting plane in 0.1mm steps. That's nearly always finer than the print resolution that will be seen in the sliced preview.

Maybe we can find a better solution if you describe more in detail why or how you need a cutting preview?

m-Th, 2017-07-10:
> A good idea, but it has one drawback: In the layers tab, you will see only the layer resolution given by the layer height setting

Use the proposed button from the **Preview tab**.

> While this might be OK for your needs, it's not optimal for every case

That's why I proposed two buttons: In **Layers** AND **Preview** tab.

> For example, when I'm using the cut feature it's nearly every time due to two cases:

Same here. :)

> Get a tiny portion of a part for a test print. In this case, it's not important if the cutting plane is +- some fractions of a mm.

Hummm... I think that here start the differences:

Theoretically this is true, but in practice there are shapes on which now it is very difficult to see where to cut for example we want the biggest width of a complex curvy object (for example a spoon).

Sebastianv650, 2017-07-10:
> Use the proposed button from the Preview tab.

Also in the Preview, you have the layer height resolution. Both tabs are containing simmilar informations.
Let's take your spoon example. With your proposed tools, you will only get the biggest width that is reached by your layer height resolution. A wider one might exist if you select a lower layer height.
That's what I mean, what you see in the preview tabs is only what you will get with exactly the settings given at this moment. Changing the setting afterwards or for fine selection, this is not enough.

m-Th, 2017-07-10:
> With your proposed tools, you will only get the biggest width that is reached by your layer height resolution.

Yes, this is what I want: if the printed object fits in the desired place or encloses exactly the desired place. I do not want to learn the mathematical / theoretical dimension(s). Just to see if my print fits.

closed 2026-09-29 by prusaslicer-bot[bot]