prusa3d/PrusaSlicer #409: Suboptimal path planing

m-Th, 2017-07-08:
### Version
Slic3r Prusa Edition
Version: 1.35.5-prusa3d-win64
Build: Slic3r-1.35.5-prusa3d-win64-full-201706151818
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 6.1.7601] 

### Behavior
When we try to print the attached STL file (generated GCode included) it gives us an suboptimal path jumping from object to object on the platter.

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
[Path-Test.zip](https://github.com/prusa3d/Slic3r/files/1132886/Path-Test.zip)

m-Th, 2017-07-08:
You can use this tool https://www.thingiverse.com/thing:44286/#files to simulate the print. (you must change the bed size to something bigger than 200x200 mm in About | Settings). Here is another case of suboptimal path chosen:
[Glow-N-311-FINAL-arranged.zip](https://github.com/prusa3d/Slic3r/files/1132910/Glow-N-311-FINAL-arranged.zip)

m-Th, 2017-08-14:
Perhaps here you can provide a "Manual" option in which we could simply drag & drop and/or Move Up / Move Down the parts from the list of the objects in the right side of GUI?

m-Th, 2017-08-16:
Here is another example which shows how bad the inter-object path planning is:

[GLOW-A504-R42.zip](https://github.com/prusa3d/Slic3r/files/1227890/GLOW-A504-R42.zip)

You can run it in the above simulator to see (STL at your disposition).

Perhaps you can get the centers of the bounding boxes of each object and run a Salesman Travel algorithm (you need it anyway) on these centers?

Use this order and after this find the smallest distance between these shapes (you have already this code isn't it?) and continue from there.

(on slightly unrelated topic, we really need the "Manual Path" feature - it is useful for sequential print also)

bubnikv, 2019-09-27:
The current master produces the following 1st layer:

when slicing as a single object
[image]

if split to separate objects (the object list shows each letter as a separate object)
[image]

I would say it is much better now. The change will be part of PrusaSlicer 2.2 release. Closing.

closed 2019-09-27 by bubnikv