prusa3d/PrusaSlicer #413: Incorrect STL interpretation

primalmotion, 2017-07-09:
### Version
1.35-5-prusa3d

### Operating system type + version
macOS 10.12.5 (16F73)

### Behavior
* Slic3r doesn't handle correctly the attached STL
* Load it in Slic3r, slice it
  * nothing special, all bare latest prusa settings,
* correct slicing
* incorrect slicing
[image]
[image]

As a note, it works perfectly in Simplify3D or Cura

#### STL/Config (.ZIP) where problem occurs
[tilescape-trial-frame-3x3.stl.zip](https://github.com/prusa3d/Slic3r/files/1133494/tilescape-trial-frame-3x3.stl.zip)

Sebastianv650, 2017-07-09:
Your model is badly corrupted. Slic3r tries to repair it, stating 5796 errors in the .stl and you can see that this try fails in your 3D preview. You can use Meshmixer to repair it, this is the result: [tilescape-trial-frame-3x3_repaired.zip](https://github.com/prusa3d/Slic3r/files/1133841/tilescape-trial-frame-3x3_repaired.zip)
No errors anymore and now it will slice.

Shit in - shit out :-p

enricoturri1966, 2018-01-18:
Fixed with https://github.com/prusa3d/Slic3r/commit/d34f3ffc8320c6929db53233cd2e17d7dac11d01

closed 2018-01-18 by enricoturri1966

FidelCapo, 2018-05-11:
Tested and fixed in Slic3rPE-1.40.0-alpha that will be released soon.