prusa3d/PrusaSlicer #414: No way to change setting duplicate_distance

CapnBry, 2017-07-09:
### Version
Slic3r 1.33.8-prusa3d-win64 (also present in master HEAD)

### Operating system type + version
Windows 10 Pro 64 bit

### Behavior
* Slic3r uses the duplicate_distance print setting to arrange objects on the build platter for the Arrange feature (and + speedbutton). The default is 6mm of space between objects which adds up to quite a lot when there are many objects on the bed. The problem, however, is that there is no way of changing this setting. It exists in [xs/src/libslic3r/PrintConfig.hpp](https://github.com/prusa3d/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.hpp#L438) but because the GUI never initializes it as an option (by calling `init_config_options()`) it gets the default value regardless of what is in the print's config ini. The only way to override this is to call the slicer from the command line, which doesn't help when using the GUI.

To fix this, duplicate_distance just needs to be added to the option list [here](https://github.com/prusa3d/Slic3r/blob/master/lib/Slic3r/GUI/Tab.pm#L570) and the user can change the setting by editing the ini file. It wouldn't be bad to add an option in the GUI to change it too, since it is already fully defined in PrintConfig.cpp. I would suggest under Output Options -> [Sequential Printing](https://github.com/prusa3d/Slic3r/blob/master/lib/Slic3r/GUI/Tab.pm#L784) as the first item?
```
            my $optgroup = $page->new_optgroup('Sequential printing');
+            $optgroup->append_single_option_line('duplicate_distance');
            $optgroup->append_single_option_line('complete_objects');
```

Sebastianv650, 2017-07-09:
> I would suggest under Output Options -> Sequential Printing as the first item?

I don't think that's the right place as it has nothing to do with output options or even only with the print settings. For me it's a default value that is independet from any profile, and most users will arrange the items by hand after setting the object count, as slic3r isn't finding a good object placement in some cases.

If you realy think it's necessary to change it from the GUI (in fact I never even thought about this value), it should go into File -> Preferences.

bubnikv, 2020-11-21:
Implemented in PrusaSlicer 2.3.0-alpha4 by clicking on the arrangement button with the right mouse button.

closed 2020-11-21 by bubnikv