prusa3d/PrusaSlicer #415: Suggestion for Preview View Change

BirkBinnard, 2017-07-09:
### Version
1.35.5

### Operating system type + version
Win10-64 Build 1703 (Creator Update)

### Behavior
Currently, when viewing a single layer in Preview mode, the Up/Down arrow keys increase or decrease the selected left or right vertical slider by one layer. This works as expected, and makes it easy to change the displayed layer by 1. However, the Left/Right arrow keys do exactly the same thing.

### Suggested Behavior
Change the function of the Left/Right arrow keys to switch between the Left/Right vertical slider. This would make it much easier to move the selected single layer up or down by 1 layer. Currently to do this requires a separate mouse click to select the active slider.

_Is this a new feature request?_
Yes

m-Th, 2017-07-11:
Agree with your idea; however, I don't think that's possible in this way. Most probably these navigation keys are hooked by the underlying widget (in this case Windows' Tracker component) and I don't think that it is a good idea to intercept them in wxWidgets.

Much easier and better (imho) would be to assign something like Ctrl+[ for the 1st tracker and Ctrl+] for the 2nd one.

BirkBinnard, 2017-07-11:
I guess things have changed (a lot) since the days when I wrote Windows apps. Back then the app code had direct control over all objects in a windows. Ctl-bracket would be a nice addition though.

closed 2017-07-11 by BirkBinnard

bubnikv, 2017-07-11:
Implemented with https://github.com/prusa3d/Slic3r/commit/3a813aaef6ff37e4159ee2d2cb4f7dc481ac7dea