prusa3d/PrusaSlicer #416: Incorrect Infill generated

BirkBinnard, 2017-07-09:
### Version
1.35.5

### Operating system type + version
Win10-64 Build 1703 (Creator Update)

### Behavior
* _Describe the problem_
Strange infill between inside and outside shells of vase shape.

* _Steps needed to reproduce the problem_
Slice attached STL file

[image]
[image]

#### STL/Config (.ZIP) where problem occurs
[STL+INI.zip](https://github.com/prusa3d/Slic3r/files/1133914/STL.INI.zip)

Sebastianv650, 2017-07-10:
That's not a bug, it's a feature! It's name is "Ensure vertical shell thickness", and you have it enabled. All this "windings" in the vase create flat angles. In worst case without this feature, you may get gaps between you perimeters from one layer to the next, resulting in a not watertight print. The thickness Slic3r will ensure is determined by the top and bottom solid layer setting and the layer height. Try reduce the number of solid layers and have a look at the changes in the preview.

BirkBinnard, 2017-07-10:
Verrryyyyyy interesting! Thanks for the explanation. Learn something every day.

closed 2017-07-10 by BirkBinnard