prusa3d/PrusaSlicer #418: Cooling fan problem Slic3r 1.35.5.16

Allram, 2017-07-12:
### Version
1.35.5.16-prusa3d-win64 - From Prusa drivers on prusa3d.com

### Operating system type + version
Windows 10 Pro 64bit

### Behavior
With this version of Sclic3r the cooling fan for the printed parts are either on or off.
It makes no difference trying to adjust the "Disbale fan for the first" ** layers, it still on before the first layer starts to print.

If i remove the checkpoint from "Keep fan always on" it's dead except for bridging purpose.

In every gcode with this version of Slic3r and with the fan on (but off for the first layer) i get this line in my gcode after startupcode:

`M106 S216.75 ; enable fan`

fucx, 2017-07-12:
I have the same issue on macOS with Slic3r 1.35.5.16-prusa3d.

The fan starts at 85% from the very beginning of the print, which is the setting for minimum fan speed in filament cooling settings. It doesn't seem to regard the "Disable fan for the first X layers" setting any more.

bubnikv, 2017-07-12:
As usual, please provide a config.ini
There are too many cooling settings, I am not sure whether it is a general
problem or a problem specific to some combination of parameters.

Allram, 2017-07-13:
Hello,

Here is my Prusa PLA 1.75mm.ini directly from Slic3r: http://textuploader.com/dkfi8

Here is my Printer .ini: http://textuploader.com/dkfe5

And here the 0.35FAST.ini: http://textuploader.com/dkfe1

bubnikv, 2017-07-13:
Thanks, I have fixed this bug with https://github.com/prusa3d/Slic3r/commit/bd93d2f334dba024e11d9f1b607c5c6d7170ec23

This bug has been introduced due to changes in the cooling logic to support different filament cooling settings for multi-material printing.

This fix will be part of the upcoming 1.36.x. Closing, please reopen if you encounter any other issue with the cooling logic.

closed 2017-07-13 by bubnikv