prusa3d/PrusaSlicer #420: Use Makefile instead of Build.PL system

CReimer, 2017-07-12:
**This isn't an issue and I didn't want to make this a pull request as it's at best too early for that.**

I did some research and tried to replace the Build.PL for the XS part of Slic3r. I don't like how cluttered the Build.PL is and I don't like that it doesn't compile multiple objects at once.

Here's what I came up with: https://github.com/CReimer/slic3r/tree/makefile

This is more or less a learning project for me. But I want to put out a few feelers early to see whether this work could be part of of the prusa3d Slic3r fork once it's ready.

If you don't like it and don't want to adopt it in any way, that's fine for me, too. I'll continue working on it to further improve my knowledge on Makefiles.

It works extremely well on Linux and at the moment it's only focussed on Linux. There's no install routine for now. As much as possible is built shared.

More and more projects abandon the Makefile system in favor of Ninja. That's what I want to experiment with next.

bubnikv, 2017-07-12:
Thank you for your effort indeed. I tried to come up with some form of Make at the beginning of my Slic3r journey and I failed, so you certainly pushed it further already.

I have couple of comments though:

- I would rather see a cmake as it could generate makefiles for various architectures, namely for the Visual Studio nmake.
- The make needs to be somehow integrated into the overall build process.
- At least the location of the boost and tbb libraries need to be configurable somehow. Currently on Windows the boost and tbb libraries are searched for at multiple "standard" locations. Either this has to be duplicated, or at least a way to override the "standard" paths shall be provided. CMake may help with this.
- The wxWidgets includes and libraries shall be referenced using the perl Alien::Wx mechanism to use the correct wxWidgets version referenced by the active Perl::Wx. This is important on Linux, where multiple wxWindows versions may get installed.
- Ideally the bundled glew, eigen, expat shall be used on Windows and OSX. 
- Bundled libadmesh shall be used on all platforms as it has been heavily modified and it will be modified further.
- The compilation parameters shall be retrieved from Config_heavy.pl by calling perl -V instead of using the hard-coded compilation parameters. This is arguable, but this is how the perl modules are normally compiled and it ensures binary compatibility of the Perl modules with the Perl interpreter.

I bet there are things I forgot to mention.

CReimer, 2017-07-12:
On 12.07.2017 17:38, bubnikv wrote:
> Thank you for your effort indeed. I tried to come up with some form of
> Make at the beginning of my Slic3r journey and I failed, so you
> certainly pushed it further already.
>
> I have couple of [...]
I thought about that, too. For multi-platform purpose there seems to be 
no way around this. Might be the next logical step for me to try.

>   * The make needs to be somehow integrated into the overall build process.
That's more or less what I meant with "there's no install routine for 
now". That's defintely something I have to take a look at.
> * At least the location of the boost and tbb libraries need to be
>     configurable somehow. Currently on Windows the boost and tbb
>     libraries are searched for at multiple "standard" locations. [...]
Yes. Boost is a little bit tricky on Linux, too. They don't provide any 
pkg-config files, which is the de facto standard on Linux to find 
libraries and their compile parameters. I know CMake helps with Boost.
>   * The wxWidgets includes and libraries shall be referenced using the
>     perl Alien::Wx mechanism to use the correct wxWidgets version
>     referenced by the active Perl::Wx. This is important on Linux, where
>     multiple wxWindows versions may get installed.
I'll take a look at that.
>   * Ideally the bundled glew, eigen, expat shall be used on Windows and OSX.
Same here.
>   * Bundled libadmesh shall be used on all platforms as it has been
>     heavily modified and it will be modified further.
I actually like that. libadmesh is not available on Arch Linux by default.
> * The compilation parameters shall be retrieved from Config_heavy.pl
>     by calling perl -V instead of using the hard-coded compilation
>     parameters. This is arguable, but this is how the perl [...]
I'm not sure if this could become a problem or not. But I'll take a look.
> 
> I bet there are things I forgot to mention.
That's enough for now 😉. It was a good idea to contact you this early. 
A lot of things I have to think about and find good solutions for.

bubnikv, 2017-07-12:
In our build scripts, I am using the following perl calls to extract config values from perl environment, namely the Config_heavy.pl:

perl -e "use Config; print \$Config{$1}"
to get ccflags, cc, ld 

perl -MExtUtils::Embed -e $1
to get perl_inc, ldopts

$path = `perl -e 'use Alien::wxWidgets; my $path=Alien::wxWidgets->shared_library_path; $path =~ s{\\\\}{/}g;`
$path =~ s{C:}{}g; print $path'`
on mingw to get the path to the wxWidgets libraries. There is some path wrangling to convert to the git bundled bash, but I suppose whether the path wrangling is necessary really depends on your CMake implementation of the build script.

On Linux and OSX, the call is simpler:
perl -e 'use Alien::wxWidgets; print Alien::wxWidgets->shared_library_path'`

On Windows perl compiled with Visual Studio, I am just using "${PERL_DIR}/site/lib/Alien/wxWidgets/msw_3_0_3_uni_cl_8/lib/"
as the Alien::wxWidgets->shared_library_path call produces something broken.

CReimer, 2017-07-12:
OK. The handwritten Makefile is gone and a bunch of CMakeLists.txt files are new. Compilation and Installation on Linux works.

I'm not sure what I should do with everything in the utils directory.

To make sure compilation on Windows is working as expected I'm going to set up a build environment on my Windows machine.

CReimer, 2017-07-13:
OK. I can't get my Windows Build Environment to work. VS2017 works and compiles now fine until I add in the necessary Include Directory for Perl.
I think Strawberry Perl is incompatible with VS2017.

bubnikv, 2017-07-13:
Strawberry Perl is certainly incompatible with visual studio.
You either have to compile everything with the strawberry perl bundled
mingw, or you have to compile everything with the same version of Visual
Studio, including the perl and its libraries.

We at Prusa Research currently build our Windows builds using Visual Studio
2013:
https://github.com/prusa3d/Slic3r/wiki/How-to-compile-Slic3r-Prusa-Edition-on-MS-Windows

CReimer, 2017-07-13:
Hmm... Manually compiling Perl helped a bit. But there's still something wrong. Could you test it with your known working environment?

CReimer, 2017-07-17:
I think everything is now in place. Compilation on Windows seems to work, except I still have problems with linking. But I still haven't tested my commits from today. It might work now.

Configuration flags are back. Everything should now behave like in Build.PL

I also added some comments into CMakeLists.txt to make things more understandable.

One last thing I still have to work on is Alien:wxWidgets. But that shouldn't be too difficult

For Windows you need to define
BOOST_ROOT
wxWidgets_ROOT_DIR
TBB_ROOT_DIR

CReimer, 2017-07-17:
OK. Just checked. All information from Alien::wxWidgets is exactly the same as what FindwxWidgets in cmake defines. So I think this is not needed.

bubnikv, 2017-08-16:
I started to work on your build system, big thanks for your effort.
Here is my initial thing, which seems to build on Visual Studio.
https://github.com/prusa3d/Slic3r/commit/5673205d2e1e144cbf630358f483f56507926d21
I have already spent two days learning CMake and bending it to compile on windows :-(

CReimer, 2017-08-16:
A really good start. I see things got a little cluttered with all the Windows "bending".

Unfortunately compilation on Linux is now broken. 
- ```/usr/bin/ld: cannot find -lc++```
There's no lc++ on linux

- ```/usr/bin/ld: cannot find -lC:wperl64dlibCOREperl524.lib```
That's an absolute path with Windows structure.

- You use ```find_package(WHATEVER)``` but never add in includes or target_link_libraries. I'm no expert in cmake, but is this actually enough?

- I'm not sure where boost-nowide is needed. It didn't need it in my branch.
If it is needed, it should be possible to dynamically link it. Especially the Debian folks are quite strict about that. And while I think about it, it might also be a good idea to add a comment, why admesh and shiny have to be compiled statically.

- A few more ```file(GLOB)``` might make things more clean. But that's just cosmetics.

A lot of negative feedback... But I think it's better to criticize now, to fix things before the next stable release.

For me it's also interesting to see where our "tactics" differ. I heavily rely on exceptions. I let it crash and check afterwards. 

```find_package(GLEW)
if (GLEW_FOUND AND NOT SLIC3R_STATIC)
```
You're trying to foresee things and handle errors before they happen.

```
if (NOT SLIC3R_STATIC)
    find_package(GLEW)
endif ()
if (NOT GLEW_FOUND)
```

bubnikv, 2017-08-17:
So I burned another day. It should compile on Windows and Linux, and the resulting XS.so will work with Perl. I am having a crash on loading on OSX though.

> You use find_package(WHATEVER) but never add in includes or target_link_libraries. I'm no expert in cmake, but is this actually enough?

Where do you see it? Maybe I have fixed that already?

> I'm not sure where boost-nowide is needed. It didn't need it in my branch.

This is a new thing, it is very much windows specifics, it is a UTF8 wrapper over Win32 system calls. The boost::nowide has been accepted into the very latest boost, but for now it is better to bundle it as it will be missing on most systems.

> If it is needed, it should be possible to dynamically link it. Especially the Debian folks are quite strict about that. 

I will link expat dynamically.

> And while I think about it, it might also be a good idea to add a comment, why admesh and shiny have to be compiled statically.

Shiny is  an intrusive profiler, and it is not compiled in by default. It looks so due to the difficulty of optional compilation with the Build.PL process, but with CMake I will just disable the Shiny files.

> A few more file(GLOB) might make things more clean. But that's just cosmetics.

Actually the CMake manual warns against the globs. It is preferred to list the source files explicitly as it reveals the dependencies. I am going to replace most the globs with a list of files.

CReimer, 2017-08-17:
In ```find_package(WHATEVER)``` WHATEVER was meant as placeholder.

There are a few find_package() in xs/CMakeLists.txt (Example: https://github.com/prusa3d/Slic3r/blob/master/xs/CMakeLists.txt#L208)

In my branch I did add include_directories() and target_link_libraries(). (Example: https://github.com/CReimer/Slic3r/blob/makefile_pr/xs/CMakeLists.txt#L145) find_package() on its own, might be enough. I'm not sure.

bubnikv, 2017-08-18:
> CMakeLists.txt#L145) find_package() on its own, might be enough. I'm not

> In my branch I did add include_directories() and target_link_libraries().
(Example: https://github.com/CReimer/Slic3r/blob/makefile_pr/xs/
CMakeLists.txt#L145) find_package() on its own, might be enough. I'm not
sure.

That is just couple of lines lower:

https://github.com/prusa3d/Slic3r/blob/952ca18bf9e84835d72838d54a5ccfb484af2fcd/xs/CMakeLists.txt#L218

CReimer, 2017-08-18:
Ohh. Errm. OK. I didn't see that.
Then there's nothing else for me to complain about :smile: 

Great work. Thanks a lot.

bubnikv, 2017-08-18:
Well, it compiles the XS shared library on Windows / Linux / OSX now. It
remains to finalize the install and test steps using CMake.
The whole exercise demanded much more of my time than I was willing to
invest. The result is longer than the original Perl based build system. The
benefits are a better control over the build dependencies (no need to
recompile the whole project as the Perl build system does not track the
dependencies reliably), and parallel building. I have played with a Ninja
"make" replacement, and it is really quick. Also there is a big benefit of
sharing the build process with the upcoming perl free command line Slic3r.

Anyway, thanks for your initial effort, it helped a lot, though I have
hoped that your solution was nearly finished :-)

CReimer, 2017-08-19:
Hmm, you mention installation. There's one bigger problem where I still couldn't find a good solution. (Mentioned here: https://github.com/prusa3d/Slic3r/pull/457#discussion_r132618221)

As far as I can see there's two possible solutions.
1. Move Slic3r.pm to somewhere where perl expects it to be.
2. Use CMake to set the right path while installing slic3r.pl

I do prefer the first one. But I'm quite sure it will break on Windows again.

I do consider my cmake branch 'nearly finished'. :smile: I just completely left everything related to windows out. :stuck_out_tongue:  Windows tends to make things unnecessary complicated.

A few other thing I still have to do in the build script for Arch Linux is.
- Place 'utils/zsh/functions/_slic3r' to '/usr/share/zsh/site-functions'
- Add a desktop file and place it in '/usr/share/applications'
- Symlink Slic3r_128px.png and Slic3r_192px.png to '/usr/share/icons/hicolor/128x128/apps/' and /usr/share/icons/hicolor/192x192/apps/

I haven't checked recently but as far as I remember Debian moves some more stuff from 'utils/*.pl' to '/usr/bin'

bubnikv, 2017-08-20:
> I do consider my cmake branch 'nearly finished'. 😄 I just completely left everything related to windows out. 😛 Windows tends to make things unnecessary complicated.

How about OSX? Linux is not the only OS in use.

It is not just the installation missing, but also running the perl unit & integration tests. These are essential in the build process. Also a manual to the build process is missing.

The build process should go like this:

Run:

BuildPL 

to install the depending Perl modules. The latest Slic3r PE installs the missing modules to local-lib as the upstream Slic3r does.

Run:
BuildPL --gui
to install the depending GUI perl modules.

On some linuxes you may hope to replace the two above by installing the particular distro packages. This should be the rule, if Slic3r is supposed to be installed by a distro package as a system-wide perl module.

Run cmake:

mk build
cd build
cmake  -DCMAKE_BUILD_TYPE=Release

make

now it is unclear to me, whether the "make" itself, or "make install" shall install slic3r XS.pm module and the XS.so (or XS.bundle on OSX or XS.dll on Windows). They both should first be installed to local-lib, then the unit & integration tests should be executed. Then first the build shall be considered sucessful.

With the XS.pm and XS.so installed in local-lib and successfully tested, one may consider installing Slic3r system-wide. Now the question is, does one want to install it as a system-wide perl module, or does one want to install it separately. Me personally, I would leave this decision to the packagers of the particular distros. For Windows, OSX and Linux AppImage, the package manager is me, and I am trying to keep everything local.

Anyway, if you want to install Slic3r as a system-wide module, you should disable the local-lib mechanism, so that the Build.PL and Build.PL --gui install the missing modules system-wide. This though is against the concept of the distro packages. You see, it is more complex than it seems.

> Place 'utils/zsh/functions/_slic3r' to '/usr/share/zsh/site-functions'
> Add a desktop file and place it in '/usr/share/applications'
> Symlink Slic3r_128px.png and Slic3r_192px.png to '/usr/share/icons/hicolor/128x128/apps/' and 
> /usr/share/icons/hicolor/192x192/apps/

I would prefer this to be part of the distro package. I think there is some distro package for arch linux, so IMHO you hould modify the distro packaging recipe.

You may learn from the debian distro packages how these things are handled by the experts.
https://packages.debian.org/sid/perl/slic3r-prusa

oh, this is yours :-)
https://aur.archlinux.org/packages/slic3r-prusa3d/

and Ubutu:
https://launchpad.net/ubuntu/+source/slic3r-prusa

and Fedora:
https://admin.fedoraproject.org/pkgdb/package/rpms/slic3r-prusa3d/

If you have any objections to the other packages, I would appreciate, if you could discuss your objections with the other package maintainers. I am no expert on linux packaging, though I worked part time for half a year in year 2000 as a package manager for SuSE.

I can image myself some "make system-wide-install" way to install the Slic3r system wide. I am surely open to proposals.

bubnikv, 2017-08-28:
I have added a "test" target to execute the perl unit & integration tests.
I have also added a post-build copy step to copy the XS.so and Slic3r/XS.pm into local lib.
https://github.com/prusa3d/Slic3r/commit/0cb491e13522e8d5847354c57bd6568e534ee733

Now one may just run
cmake ..
make
make test
or
ctest --verbose
to get a full output of the tests

After the "make" command the Slic3r XS.so and Slic3r/XS.pm are installed into the local-lib directory, and slic3r could be started.

I did not touch your install, but IMHO it may make sense to have two kinds of installs:
1) A system-wide perl intrusive install
2) A kind of a system-wide install, but the Slic3r modules installed somewhere locally to Slic3r outside of the system-wide perl.

Now someone should write a documentation to the build process. Any volunteer?

CReimer, 2017-08-29:
On 28.08.2017 19:29, bubnikv wrote:
> I did not touch your install, but IMHO it may make sense to have two 
> kinds of installs:
> 
>  1. A system-wide perl intrusive install

I already thought about that and tried to implement it. But I cannot 
find a way to easily get Perl's module path without /usr (or /usr/local) 
prefix. But to keep all functions within cmake working I need it without 
this prefix. It's cmake job to add in the prefix.
The more difficult variant here is to get the path with prefix from 
Perl, get the prefix Perl was compiled with and use cmake's string 
modification features to remove it and readd the prefix cmake want's to use.
FindPerl or FindPerlLibs don't currently provide the prefix which perl 
was compiled with. But that's possible with a small additional cmake 
module and some borrowed code from FindPerlLibs

>  2. A kind of a system-wide install, but the Slic3r modules installed
>     somewhere locally to Slic3r outside of the system-wide perl.

That's more or less the case right now. But even there's a problem with 
the prefix. slic3r.pl relies on FindBin to find /usr/bin (or 
/usr/local/bin). Slic3r's perl modules are in /usr/lib/slic3r-prusa3d 
(or /usr/local/.....). To work around this we could use

```
use lib "$FindBin::Bin/../lib/slic3r-prusa3d
```
as suggested by @jkoelker

> 
> Now someone should write a documentation to the build process. Any 
> volunteer?
Well, eeermm. No thank you. :D

But is there really much necessary?

```
mkdir build
cd build
cmake ..
make
make install
```
Ohh I forgot about Windows again...

bubnikv, 2017-08-30:
I have removed the xs/Build.PL and I am using the new process at our build server.

I think we can close this issue as of now. Let's keep the fingers crossed.

bubnikv, 2017-08-30:
I see I have forgotten the installation issue, so let's keep the issue open.

CReimer, 2017-08-30:
I just checked with the default perl on Arch Linux. For a system-wide installation there's no need to check for the prefix. Perl doesn't allow modules in /usr/local/lib/perl5 only in /usr/lib/...

The PERL_VENDORLIB from FindPerlLibs.cmake should be enough then.

bubnikv, 2017-08-31:
There is a possibility of installing the Slic3r with the Slic3r libraries
(the XS.so and the Slic3r Perl Modules) out of the Perl default paths. This
makes a lot of sense, if one wants to have both the mainline and the Slic3r
Prusa Edition installed on the same machine.

If you open our binary distribution package, you will find the following
Slic3r bash script:

#!/bin/bash
DIR=$(dirname "$0")
export LD_LIBRARY_PATH="$DIR/bin"
export PERL5LIB=
export PERLLIB=
exec "$DIR/bin/perl5.22.0" "$DIR/bin/slic3r.pl" "$@"

This is to let the bundled perl use a local PERL5LIB relative to the
bundled perl binary.

Now if you want to force the Slic3r modules to be installed outside the
system Perl space to be able to mix the two slic3rs, you may just copy the
content of src-perl/Slic3r/local-lib/lib to /usr/lib/Slic3r-PrusaEdition,
install the slic3r.pl to /usr/lib/Slic3r-PrusaEdition/bin/
and install a following starter script to /usr/bin

#!/bin/bash
export PERL5LIB=/usr/lib/Slic3r-PrusaEdition:${PERL5LIB}
exec perl "/usr/lib/Slic3r-PrusaEdition/bin/slic3r.pl" "$@"

That should do in my opinion and it should make the packagers happy as well.

CReimer, 2017-09-04:
To make a system-wide installation possible without conflict with mainline Slic3r you could also just rename the perl modules to something like Slic3r-P3D.

While updating the package script for Arch Linux I ran into another problem. The tests don't work on linux like that. At least not on Arch Linux. Because you expect 'prove' to be in ${PERL_BIN_PATH}. Unfortunately it's not. Prove is in ${PERL_BIN_PATH}/core_perl. And this path is part of the PATH environment variable. Just calling 'prove' works.

lordofhyphens, 2017-10-02:
Slic3r uses local-lib so its installs are safe side-by-side.

bubnikv, 2019-01-07:
@vojtechkral is polishing a new build system for 1.42 PE, which will implement install.
Closing this one.

closed 2019-01-07 by bubnikv