prusa3d/PrusaSlicer #421: LCD Speed Modifier Lost on Tool Change

Ziethriel, 2017-07-12:
Version:
1.35.5.16

Operating system:
Windows7 Pro SP1

Behavior:
Currently a manual speed change made via the LCD during a MMU print is not preserved when making a tool change. I believe that a speed change should persist from tool to tool as the speed percent on the LCD should be a global modifier, more specific tweaks for different filaments and whatnot should be made in the slicer.

Steps needed to reproduce the problem:
1) Slice any multi material print
2) Begin Printing Normally
3) Make speed % adjustment on LCD controller
4) Observe as adjustment is removed when changing tools.

Is this a new feature request?
No

bubnikv, 2017-07-12:
The filament exchange sequence is sensitive to speeds and timing, so it is generally not a good idea to figet with the print speed override to maintain a reliable filament switching.

In addition, based on the filament to be loaded, Slic3r may lower the print speed using the G-code
M220 - Set Feedrate Percentage
Namely, Slic3r lowers extrusion speed for soluble materials like PVA or E3D scaffold.

The Slic3r wipe tower is based on the implementation by @akukan, I have to check with him what were his reasons to emit the M220 code to slow down PVA or E3D scaffold, as these soluble materials are usually only used for support, and the support print speed could be configured in Slic3r separately.

bubnikv, 2017-07-12:
Also there is no way to temporarily override the feed rate override in the Marlin firmware, so if we wanted to disable your manually entered feed rate override just for the filament exchange sequence, there is no way to set it back to your value. It seems to me that to implement your request would require us to extend the firmware in a weird way to stash your feed rate override value somewhere into firmware, then do the filament exchange procedure, and then restore your feed rate override from the stash. I am not quite sure whether it is worth the effort.

Ziethriel, 2017-07-12:
So I don't write firmware, so I could be up my own *** here, but you have all these values like speed, flow, nozzle, bed, fan speed, and mode. Couldn't a hidden value "restore speed" be added there that the gcode references in restoring speed when the wipe tower is complete?

I'm sure there are more important things to be done, but adjusting speed and flow (less important here) on the fly are very useful features to me before the MMU. I'm often asked to get a part done for a meeting that was originally due at the end of the day and being able to bump up a 4 hour print by 25% is super helpful.

bubnikv, 2017-07-14:
I have discussed with @akukan. We would like to fulfill your request in a systematic way, which requires more effort than it seems.

1) Extend the firmware with a new G-code or M-code to stash the user feed rate override, and to restore it on demand. Or maybe yet better to suppress the user override.

2) Extend slic3r to support per filament speed and acceleration settings.

As of now, this is postponed, but it will be planned.

bubnikv, 2018-12-12:
According to Pavel Sindler, our FW lead, there will be 

M220 B - backup
M220 R - restore

codes supported by the Prusa MMU firmwares.

We should integrate them.

bubnikv, 2018-12-17:
Implemented with d0edc3640062316d8a137a41387e00395b2ef1ae

The current logic of the usage of the M220 is as follows:

If a flex, PVA or BVOH filament is used together with the usual filaments, then the behavior is unchanged, because the wipe tower generator issues M220 to slow down the print on these filaments, and it issues M220 to recover the print speed to 100% for the usual filaments.

If only the usual filaments are printed with, then the wipe tower generator issues the M220 to set the print speed to 100% at the wipe tower, also it issues M220 B to back up the old print speed override, and it issues M220 R to restore the old print override before leaving the wipe tower.

closed 2018-12-17 by bubnikv