prusa3d/PrusaSlicer #422: Only bottom 4 layers are sliced

jangrewe, 2017-07-12:
### Version
1.35.5 / x64

### Operating system type + version
Windows 10

### Behavior
I am trying to slice an STL with 10.8mm height, but the resulting G-Code only contains the first 4 layers (0.8mm). The 3D view shows the whole part, but in the preview all of the upper layers are already missing.

3D View: https://i.snag.gy/1Ldp24.jpg
Preview: https://i.snag.gy/7wvaY5.jpg

#### STL/Config (.ZIP) where problem occurs
STL/G-Code: [Beeper.zip](https://github.com/prusa3d/Slic3r/files/1142791/Beeper.zip)

I am using the current commit of the Slic3r Settings repo (MK2 MM folder).

bubnikv, 2017-07-12:
The wall is too thin to fit two perimeters. You have either to make the
wall thicker, or to enable the "detect thin walls" at the Print Settings ->
Layers and perimeters page.
Then it slices fine.

jangrewe, 2017-07-12:
Thanks, at 1.2mm width it worked fine.

closed 2017-07-12 by jangrewe