prusa3d/PrusaSlicer #423: Internally facing perimeters

pcsentinel, 2017-07-16:
### Version
prusa 1.35.5

If you print a ring i.e. an 'O' shape, with perimeter printing set to not external first, the outer edge of the ring is printed correctly, but the inner perimeters are printed with the innermost, i.e. the perimeter that is closest to the geometric center of the ring printed first. Shouldn't the inside of a ring be treated as an external perimeter?

bubnikv, 2017-07-20:
I am not sure whether I understand you correctly.

The way it works, the outer contour of the outer ring and the innermost contour of the inner ring are considered to be the external perimeters, and they are printed before the perimeters they are touching are printed. In my opinion it works as intended.

pcsentinel, 2017-07-20:
I don't seem to get that, please see attached, the external perims of the outside print last, but for the internal perims the perimeter that is furthest from the centre of the object prints last.

[Drawing2.zip](https://github.com/prusa3d/Slic3r/files/1163356/Drawing2.zip)

bubnikv, 2017-07-21:
> I don't seem to get that, please see attached, the external perims of the outside print last, but for the internal perims the perimeter that is furthest from the centre of the object prints last.

I don't understand your words either.

> for the internal perims the perimeter that is furthest from the centre of the object prints last.

When running your and mine G-codes through simplify3d g-code visualizer, I see the perimeters to be printed in the following order:
[image]

and I consider that to be correct. Do you think this order is not correct?

bubnikv, 2017-07-21:
I believe it is working as intended. Closing.

closed 2017-07-21 by bubnikv