prusa3d/PrusaSlicer #425: Preconfigure for i3 MK2

probonopd, 2017-07-16:
### Version
Slic3r-1.35.5-prusa3d-linux64-full-201706151818.AppImage

### Operating system type + version
openSUSE LEAP 42.2

### Behavior
* _Describe the problem_
* Download the AppImage and run it for the first time. It will ask very basic questions, such as the size of the bed
* _Expected Results_
*  would expect this to be preconfigured, and select me "i3 MK2" from a dropdown menu
* _Actual Results_
* Despite this being the "Prusa Edition", it seems to lack any kind of presets for Prusa machines

_Is this a new feature request?_
Yes

Settings are at https://github.com/prusa3d/Slic3r-settings/tree/master/Slic3r%20settings%20MK2 but the user has to copy these manually to `$HOME/.Slic3r/`. It would be nice if the software would copy them there automatically, after asking which printer one has.

bubnikv, 2017-07-17:
We are planning to do that in some form.

probonopd, 2017-09-21:
Implemented it as a Perl script that runs before the actual Slic3r application:

[image]

Place the contents of https://github.com/prusa3d/Slic3r-settings in `usr/share/slic3r/settings/` and this script in `usr/bin/preconfig.pl` inside the AppDir:

```
#!/usr/bin/env perl

# Slic3r settings selection dialog for use with AppImage
# by Simon Peter <probono@puredarwin.org>

# This script is licensed under the GNU Affero General Public License, version 3
# 
# This script will ask the user to select one of the
# Slic3r profiles for the 3D printers produced by Prusa Research
# when the application is started for the first time and no
# existing Slic3r configuration exists
# https://github.com/prusa3d/Slic3r/issues/425
#
# For this to work, the content of https://github.com/prusa3d/Slic3r-settings
# needs to be bundled at (directory of this script)/../../share/slic3r/settings/
# and this script must be invoked prior to running Slic3r

use Wx 0.15;
use strict;

package MyFrame;

use Wx qw[:everything];
use base qw(Wx::Frame);
use strict;
use File::Basename;
# use File::Copy::Recursive qw(dircopy);

use Wx::Locale gettext => '_T';
sub new {
    my( $self, $parent, $id, $title, $pos, $size, $style, $name ) = @_;
    $parent = undef              unless defined $parent;
    $id     = -1                 unless defined $id;
    $title  = ""                 unless defined $title;
    $pos    = wxDefaultPosition  unless defined $pos;
    $size   = wxDefaultSize      unless defined $size;
    $name   = ""                 unless defined $name;

    $self = $self->SUPER::new( $parent, $id, $title, $pos, $size, $style, $name );
    $self->{label_1} = Wx::StaticText->new($self, wxID_ANY, _T("Please select your printer"));
    $self->{combo_box_1} = Wx::ComboBox->new($self, wxID_ANY, "", wxDefaultPosition, wxDefaultSize, [_T("Original Prusa i3 MK2 & MK2S"), _T("Original Prusa i3 MK2 Multi Material"), _T("Prusa i3 with a 1.75mm filament extruder"), _T("Prusa i3 with a 3mm filament extruder")], wxCB_DROPDOWN);
    $self->{button_1} = Wx::Button->new($self, wxID_OK, "");

    $self->SetTitle(_T("Select printer"));
    $self->{combo_box_1}->SetSelection(-1);

    $self->{sizer_1} = Wx::BoxSizer->new(wxVERTICAL);
    $self->{sizer_2} = Wx::BoxSizer->new(wxHORIZONTAL);
    $self->{sizer_2}->Add($self->{label_1}, 0, 0, 0);
    $self->{sizer_1}->Add($self->{sizer_2}, 1, 0, 0);
    $self->{sizer_1}->Add($self->{combo_box_1}, 0, 0, 0);
    $self->{sizer_1}->Add($self->{button_1}, 0, 0, 0);
    $self->SetSizer($self->{sizer_1});
    $self->{sizer_1}->Fit($self);
    $self->Layout();
    $self->Show(1);

    Wx::Event::EVT_BUTTON($self, $self->{button_1}->GetId, \&OkClicked);

    return $self;

}

sub OkClicked {
    my $self = shift;
    my $selection = $self->{combo_box_1}->GetSelection();
    print "$selection\n";
    my $path = "";

    if ($selection == 0) {
        $path = "Slic3r settings MK2";
    } elsif ($selection == 1) {
        $path = "Slic3r settings MK2MM";
    } elsif ($selection == 2) {
        $path = "Slic3r settings 175";
    } elsif ($selection == 3) {
        $path = "Slic3r settings 3mm";
    } else {
        print "Invalid selection\n";
        $self->Close();
    }

    if ($path ne "") {
        my $dirname = dirname(dirname(dirname(__FILE__))) . "/share/slic3r/settings/" . $path;
        print("$dirname\n");

        if ( -d $dirname ) {
            print "Copying $dirname to ~/.Slic3r/\n" ;
            # dircopy($dirname, $ENV{"HOME"} . "/.Slic3r/" );
            my @args = ("cp", "-r", $dirname, $ENV{"HOME"} . "/.Slic3r/");
            system(@args) == 0
                or die "system @args failed: $?";

        } else {
            print "$dirname is missing, cannot copy to ~/.Slic3r/\n" ;
        }
    }
    $self->Close();
}

# end of class MyFrame

1;

package main;

unless(caller){

    if ( -d $ENV{"HOME"} . "/.Slic3r/" ) {
        # Do nothing since Slic3r is already configured
        print "~/.Slic3r/ already exists, skipping printer selection dialog\n" ;
    } else {
        # Show the printer selection dialog GUI
        local *Wx::App::OnInit = sub{1};
        my $app = Wx::App->new();
        Wx::InitAllImageHandlers();
        my $frame_1 = MyFrame->new();
        $app->SetTopWindow($frame_1);
        $frame_1->Show(1);
        $app->MainLoop();
    }
}
```

Finally, add a line that calls the script in `usr/bin/slic3r` like this:

```
#!/bin/bash
DIR=$(dirname "$0")
export LD_LIBRARY_PATH="$DIR/bin"
export PERL5LIB=
export PERLLIB=
exec "$DIR/bin/perl5.22.0" "$DIR/bin/preconfig.pl"
exec "$DIR/bin/perl5.22.0" "$DIR/bin/slic3r.pl" "$@"
```

probonopd, 2017-11-03:
Ping @bubnikv

bubnikv, 2017-11-13:
Simon,

Sorry for wasting your time, but we went into another direction. The upcoming Slic3r will have the "compatible printers" feature ported from the mainline Slic3r, allowing one to filter out the print and filament profiles, which are not compatible with the printer settings. You may already play with it at the CPP_presets branch, where the presets were reworked from Perl to C++. At the end it will work very much the same way as the Prusa Control.

Vojtech

bubnikv, 2017-12-10:
I have bent the configuration wizard to offer a selection of bundled config_bundle files. If a config bundle is selected, then it is unpacked into user's profile. If a generic "Other" is selected, then the usual config wizard workflow follows. This allows us to accept configuration bundles from other vendors as well.
Thanks for your inspiration anyway.

bubnikv, 2017-12-10:
The config wizard has been extended with https://github.com/prusa3d/Slic3r/commit/657f2734f19da1d6483b9d70b5d8102e727bc34b
A first config bundle with the Original Prusa i3 MK2 & MK3 profiles has been added with https://github.com/prusa3d/Slic3r/commit/98fdb46da9741a4624a427b00ef355b7ee25abc6

We will likely add one profile for a "Generic i3".

Screenshots follow:

[image]

[image]

[image]

bubnikv, 2017-12-14:
Implemented in 1.38.4. Closing.

closed 2017-12-14 by bubnikv