prusa3d/PrusaSlicer #426: Wipe Tower ... only supported ... .4 nozzle error when slicing a non .4mm nozzle single extruder (Stock MK2S) print

devsfan1830, 2017-07-17:
### Version
1.35.5.16

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 Pro (Build 15063.483)

### Behavior
Slic3r refuses to slice anything when the nozzle diameter is anything other than 0.40mm.

Load any STL, select Original Prusa i3 MK2 as the "Printer" and Under Extruder 1 in Printer Settings change the Nozzle diameter to any value other than 0.40. Click "slice now". The model will not slice. The bottom left of the window says "The Wipe Tower is currently only supported for the 0.4mm nozzle diameter."

The same occurs with the Original Prusa i3 MK2 0.25 Nozzle with no changes. Just click "Slice now". The model will still fail to slice.

At no point have I selected the MM printer preset. I cannot locate any "Wipe Tower" related settings to overcome this error.

#### STL/Config (.ZIP) where problem occurs
[configs.zip](https://github.com/prusa3d/Slic3r/files/1151501/configs.zip)

bubnikv, 2017-07-17:
You get the message when the print profile has the wipe tower enabled at the "Multiple Extruders" page. Once you uncheck the checkbox, you will be good.

I will loosen the constraint, so you will get this error message only if the printer settings has more than one extruder configured, as the wipe tower does not get generated for a single extruder print.

bubnikv, 2017-07-17:
Fixed with https://github.com/prusa3d/Slic3r/commit/2c82a327dd9e5270cf0f3a976360c8ec99ed3452

closed 2017-07-17 by bubnikv

devsfan1830, 2017-07-17:
Awesome thanks! I think another simpler alternative would have been to remove the default checkbox for the wipe tower enable in the non-MM printer presets. Maybe I was not cleat n my initial post. The default printer profiles for MK2 and Mk2 .25 Nozzle have the wipe tower turned on by default on a clear install..

bubnikv, 2017-07-18:
> I think another simpler alternative would have been to remove the default
checkbox for the wipe tower enable in the non-MM printer presets. Maybe I
was not cleat n my initial post. The default printer profiles for MK2 and
Mk2 .25 Nozzle have the wipe tower turned on by default on a clear install..

It is all done on purpose, as there is not a link between the print and
printer profile, so the print configuration does not know about whether the
printer has one or more extruders. It should be improved, I know, but this
is what we have as of now.