prusa3d/PrusaSlicer #427: Fill angle of any number other than an Integer does not work

geodave810, 2017-07-17:
### Version
_Version of Slic3r used goes here_ 1.36.0

_Use `About->About Slic3r` for release versions_1.36.0-prusa3d-win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_Windows 7 Pro Build 7601: Service Pack 1

### Behavior
* _Describe the problem_Fill angle of any number other than an Integer does not work.
* _Steps needed to reproduce the problem_ Under Print Settings/Infill/Advanced/Fill Angle set it to 22.5
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_ I expect fill pattern of 22.5 degrees
* _Actual Results_ I get no fill pattern
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

_Is this a new feature request?_No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[Cube_Test.zip](https://github.com/prusa3d/Slic3r/files/1153729/Cube_Test.zip)

bubnikv, 2017-07-18:
As of now, the angle values stored into the internal config storage are
integers. This applies to support material angle as well.
We should at least avoid the user to reject the non-integer angle values.

What is your use case to enter non-integer infill angle values?

geodave810, 2017-07-18:
If I had an object that was mostly located 67.5 degrees on the print bed like shown in the attached image, I would use a fill angle of 22.5 degrees.  My work around has been to use the next integer value up or down, 23 or 22 degrees.  Rejecting non-integer angles would be sufficient for me.  I have attached an image and STL file as examples.
[image]

[Cube_67_5_Test.zip](https://github.com/prusa3d/Slic3r/files/1155613/Cube_67_5_Test.zip)

bubnikv, 2017-07-20:
Fixed with https://github.com/prusa3d/Slic3r/commit/969e3f4a80552a2fce08537c34250d6d34135e08
Enabled floating rotation angle for both infill and supports.

closed 2017-07-20 by bubnikv

geodave810, 2017-07-20:
Cool. Thanks.