prusa3d/PrusaSlicer #428: GUI properties panel "broken" after #377 fix

anton-veretenenko, 2017-07-17:
1.36.0-prusa3d-linux64
Linux Ubuntu 16.04.2

There is an update introduced in a7153c6 (#377 issue).
And this update broke properties panel for me. It got narrower than it was before and even got a scroll bar at the bottom.
Screenshot 1 shows how it was in 1.35.5 version:
[image]

Screenshot 2 shows how it is in 1.36.0 version:
[image]

bubnikv, 2017-07-18:
Thanks for heads up. The various platforms behave different. I would have to put some limit on the minimum width on Linux I suppose. That's the beauty of multi-platform programming.

bubnikv, 2017-07-20:
Should be fixed with https://github.com/prusa3d/Slic3r/commit/7103fa78ff732c2b8a23ec83937740119fc4044c

closed 2017-07-20 by bubnikv

anton-veretenenko, 2017-08-04:
Unfortunately I still see the same layout with scroll bar at the bottom with 1.36.2 version.
Properties panel is wider but still does not fit.