prusa3d/PrusaSlicer #429: Parts of "Settings for (part).stl" dialog not working

BirkBinnard, 2017-07-19:
### Version
1.36.0

### Operating system type + version
Win10=64 Creators UPdate

### Behavior
Objects in the Move box of the attached dialog are not selectable and consequently don't work. 

[image]

What I'd like to do is lower the part below the print surface (by using the Z: slider or text box) so only some of the 2 vertical posts actually prints.

_Is this a new feature request?_

No - request to fix bug

Sebastianv650, 2017-07-19:
Use the cut feature for this, it will do what you want.

BirkBinnard, 2017-07-19:
Thanks Sebastian, I discovered this myself about 15 minutes after I made my original post. You are correct of course, the Cut feature worked fine and I was able to do the test I wanted.
But....is there some reason the controls in the Move box don't work? Using them would be a lot more intuitive and straightforward. Here's hoping they get fixed in some future update.

bubnikv, 2017-07-19:
I am not sure myself how to handle the UI.

The sliders are targeted to the modifier meshes, they were implemented by
@lordofhyphens to move the "generic" modifier meshes around. I am not sure
how useful these sliders would be for the object volumes, as you would not
be able to align them with these sliders anyway.

BirkBinnard, 2017-07-19:
Thanks for your feedback, I do appreciate it. I haven't looked at the code involved, so it is not clear to me what modifier meshes and object volumes are, but from an end-user point of view I'd expect the sliders & text boxes to simply change the logical location of the selected object by the X, Y, and Z values entered using either the respective sliders or text boxes.

lordofhyphens, 2017-07-19:
At one point it did move parts around, at least in mainline slic3r

bubnikv, 2018-12-11:
We have reworked the UI for object and volume manipulation, it will be part of the upcoming 1.42 alpha release. Closing.

closed 2018-12-11 by bubnikv