prusa3d/PrusaSlicer #430: Unnecessary raft on models with large perimeter but nothing inside.

JakeAllison, 2017-07-19:
Version 1.36.0-prusa3d-linux64

Ubuntu Studio 16.04.2 LTS
4.8.0-58-lowlatency

This is a feature request.

The issues is that when I make a raft on an object with a large perimeter but nothing in the middle, the raft will cover all the area within and waste large amounts of filament and time.

The screen shot attached shows all the excess raft  that is going to waste. It would be nice to have the raft only cover the areas that are necessary.

[image]

bubnikv, 2017-07-19:
Actually Slic3r fills the holes with the raft on purpose to improve 1st
layer adhesion. It may be worth though to fill the holes at the 1st layer
only.

Would you please attach your STL & config.ini, so I can reproduce your case?

JakeAllison, 2017-07-20:
Here's the files.

The configs used are:
Print Settings: Raft
Filament Settings: Colorfabb PLA
Printer Settings: Creator Pro Right Only

[Phantom_3_Prop_Guards.zip](https://github.com/prusa3d/Slic3r/files/1163822/Phantom_3_Prop_Guards.zip)
[Slic3r_Pr_config_bundle.ini.zip](https://github.com/prusa3d/Slic3r/files/1163825/Slic3r_Pr_config_bundle.ini.zip)

bubnikv, 2021-02-24:
[image]

Fixed with e9875f57a9cdcd32d6244dd38200078f5c2a905c

    Don't fill in the holes in 1st layer of the raft.
    The user may apply a higher raft_expansion if one wants a better 1st layer adhesion.

closed 2021-02-24 by bubnikv

ymralin, 2022-11-21:
Is it possible to bring back this feature as an option? I know there is Raft expansion value, but it expands in all directions. A continous raft under whole model would be helpfull. Another helpfull thing here would be to force the raft to be printed contionusly, which now is no the case when the model base is an unregular concave polygon.