prusa3d/PrusaSlicer #431: Error building version_1.36.0 on debian linux

mavu, 2017-07-20:
### Version
version_1.36.0

### Operating system type + version
Debian testing

### Behavior

`t/avoid_crossing_perimeters.t .. TBB Warning: Exact exception propagation is requested by application but the linked library is built without support for it
Caught C++ exception of type or derived from 'std::exception': std::bad_alloc at /home/mavu/repo/Slic3r/t/../lib/Slic3r/Print/Object.pm line 192.
# Looks like your test exited with 2 before it could output anything.
t/avoid_crossing_perimeters.t .. Dubious, test returned 2 (wstat 512, 0x200)
Failed 1/1 subtests 
t/bridges.t .................... 1/16 TBB Warning: Exact exception propagation is requested by application but the linked library is built without support for it
Caught C++ exception of type or derived from 'std::exception': std::bad_alloc at /home/mavu/repo/Slic3r/t/../lib/Slic3r/Print/Object.pm line 192.
# Looks like your test exited with 2 just after 14.
t/bridges.t .................... Dubious, test returned 2 (wstat 512, 0x200)
Failed 2/16 subtests 
t/clean_polylines.t ............ ok   
t/clipper.t .................... ok   
t/collinear.t .................. ok     
t/combineinfill.t .............. TBB Warning: Exact exception propagation is requested by application but the linked library is built without support for it
Caught C++ exception of type or derived from 'std::exception': std::bad_alloc at /home/mavu/repo/Slic3r/t/../lib/Slic3r/Print/Object.pm line 192.
# Looks like your test exited with 2 before it could output anything.
t/combineinfill.t .............. Dubious, test returned 2 (wstat 512, 0x200)
Failed 8/8 subtests 
t/config.t ..................... ok   
t/cooling.t .................... 1/15 TBB Warning: Exact exception propagation is requested by application but the linked library is built without support for it
Caught C++ exception of type or derived from 'std::exception': std::bad_alloc at /home/mavu/repo/Slic3r/t/../lib/Slic3r/Print/Object.pm line 192.
# Looks like your test exited with 2 just after 10.
t/cooling.t .................... Dubious, test returned 2 (wstat 512, 0x200)
Failed 5/15 subtests 
t/custom_gcode.t ............... TBB Warning: Exact exception propagation is requested by application but the linked library is built without support for it
Caught C++ exception of type or derived from 'std::exception': std::bad_alloc at /home/mavu/repo/Slic3r/t/../lib/Slic3r/Print/Object.pm line 192.
# Looks like your test exited with 2 before it could output anything.
t/custom_gcode.t ............... Dubious, test returned 2 (wstat 512, 0x200)
Failed 15/15 subtests 
t/dynamic.t .................... skipped: variable-width paths are currently disabled
t/fill.t ....................... 1/19 TBB Warning: Exact exception propagation is requested by application but the linked library is built without support for it
Caught C++ exception of type or derived from 'std::exception': std::bad_alloc at /home/mavu/repo/Slic3r/t/../lib/Slic3r/Print/Object.pm line 192.
# Looks like your test exited with 2 just after 7.
t/fill.t ....................... Dubious, test returned 2 (wstat 512, 0x200)
Failed 12/19 subtests 
	(less 1 skipped subtest: 6 okay)
t/flow.t ....................... TBB Warning: Exact exception propagation is requested by application but the linked library is built without support for it
Caught C++ exception of type or derived from 'std::exception': std::bad_alloc at /home/mavu/repo/Slic3r/t/../lib/Slic3r/Print/Object.pm line 192.
# Looks like your test exited with 2 before it could output anything.
t/flow.t ....................... Dubious, test returned 2 (wstat 512, 0x200)
Failed 6/6 subtests 
t/gaps.t ....................... TBB Warning: Exact exception propagation is requested by application but the linked library is built without support for it
Caught C++ exception of type or derived from 'std::exception': std::bad_alloc at /home/mavu/repo/Slic3r/t/../lib/Slic3r/Print/Object.pm line 192.
# Looks like your test exited with 2 before it could output anything.
t/gaps.t ....................... Dubious, test returned 2 (wstat 512, 0x200)
Failed 1/1 subtests 
t/gcode.t ...................... 1/25 TBB Warning: Exact exception propagation is requested by application but the linked library is built without support for it
Caught C++ exception of type or derived from 'std::exception': std::bad_alloc at /home/mavu/repo/Slic3r/t/../lib/Slic3r/Print/Object.pm line 192.
# Looks like your test exited with 2 just after 1.
t/gcode.t ...................... Dubious, test returned 2 (wstat 512, 0x200)
Failed 24/25 subtests 
t/geometry.t ................... ok     
t/layers.t ..................... TBB Warning: Exact exception propagation is requested by application but the linked library is built without support for it
Caught C++ exception of type or derived from 'std::exception': std::bad_alloc at /home/mavu/repo/Slic3r/t/../lib/Slic3r/Print/Object.pm line 192.
# Looks like your test exited with 2 before it could output anything.
t/layers.t ..................... Dubious, test returned 2 (wstat 512, 0x200)
Failed 5/5 subtests 
t/loops.t ...................... skipped: temporarily disabled
t/multi.t ...................... TBB Warning: Exact exception propagation is requested by application but the linked library is built without support for it
Caught C++ exception of type or derived from 'std::exception': std::bad_alloc at /home/mavu/repo/Slic3r/t/../lib/Slic3r/Print/Object.pm line 192.
# Looks like your test exited with 2 before it could output anything.
t/multi.t ...................... Dubious, test returned 2 (wstat 512, 0x200)
[... 447 lines cut ...]
  Parse errors: Bad plan.  You planned 13 tests but ran 0.
t/perimeters.t               (Wstat: 512 Tests: 52 Failed: 1)
  Failed test:  52
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 59 tests but ran 52.
t/print.t                    (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 6 tests but ran 0.
t/retraction.t               (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 26 tests but ran 0.
t/shells.t                   (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 21 tests but ran 0.
t/skirt_brim.t               (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 6 tests but ran 0.
t/threads.t                  (Wstat: 6 Tests: 2 Failed: 1)
  Failed test:  2
  Non-zero wait status: 6
Files=30, Tests=229, 208 wallclock secs ( 0.08 usr  0.04 sys + 756.73 cusr  0.74 csys = 757.59 CPU)
Result: FAIL
Some tests failed. Please report the failure to the author!

`

bubnikv, 2017-07-20:
Thanks for being vigilant. I am aware of the issue, I was too tired yesterday evening to fix it.

mavu, 2017-07-20:
You're welcome, it made me update my system, which i had put off for too long anyway :)

bubnikv, 2017-07-20:
Fixed with https://github.com/prusa3d/Slic3r/commit/aaefb768880c40c7230590240bf4b058427b1e27

closed 2017-07-20 by bubnikv