prusa3d/PrusaSlicer #434: Different per color change depth for different extruders for the wipe tower

Mr-Russian, 2017-07-21:
### Version
Version 1.36.0-prusa3d

### Operating system type + version
Mac OS Sierra 10.12.6

### Behavior
I think it would a very nice feature to be able to select a different per color change depth on the wipe tower for different extruders. For example, I want to print a dual color object with black and white filament. As we all know, white filament will not come out clean after the black filament but will continue being grayish for a while, while black filament will quickly come out black and easily clean out the white filament. To save filament and printing time, it would be useful to set different color change depths for different filaments on the wipe tower.

_Is this a new feature request?_ YES

bubnikv, 2017-07-21:
Yet better, you would have the change depth configured per color1 -> color2 pair.

Mr-Russian, 2017-07-22:
Yes so even if you have 3 or 4 colors and only one of the colors needs long wipe to become clear you can have the option of doing that. The other colors could have their own change depth which could be shorter.