prusa3d/PrusaSlicer #435: Filament "other layer" settings

sarf2k4, 2017-07-21:
I am currently using the 1.36.0 on win10 64 bit, I was printing and often set the the "other layer" setting to 0 in filament settings tab. To my surprise that a new gcode line were added that is to turn off the bed temperature to 0. Was this intended or a bug in this version?

As far as I remember it, putting 0 will not put a new gcode into the gcode file. Care to take a look on this one?

bubnikv, 2017-07-21:
I rewrote the G-code generator and I also rewrote the cooling logic. For example, Slic3r did not start the fan over bridges when "enable auto cooling" was off, which I consider to be a bug, as the fan over bridge should be always configurable as the UI already hints by enabling / disabling the UI fields.

I also consider it reasonable to enable the heat bed for the 1st layer only for some materials, and then to be able to switch the bed heater off for the other layers, so the current implementation seems to be reasonable. I should adjust the tooltips though.

rtyr, 2020-03-22:
I also believe that current implementation is reasonable. Tooltips are fixed. Closing.

Dromantor, 2021-04-23:
"Tooltips are fixed." Sure about that?
[image]

rtyr, 2021-04-23:
No, you are right. Tooltips are not fixed.

closed 2026-09-29 by prusaslicer-bot[bot]