prusa3d/PrusaSlicer #436: Add message saying "slice first before preview" to Slic3r 1.35.5.16

WhiteDragonD, 2017-07-21:
### Version
Slic3r 1.35.5.16

### Operating system type + version
Windows 7

### Behavior
In the previous version of Slic3r, it would auto slice so when going to the preview tab, it would show.  After updating, the app isn't auto slicing so I thought it was broken because I couldn't see the layers in the preview.

Request to add a messaging saying something like "Please slice the object before previewing layers" so we know it hasn't been sliced yet.

#### STL/Config (.ZIP) where problem occurs
NA

Sebastianv650, 2017-07-21:
This message box would need a "Don't show next time" check box. Maybe it would even better if the slicing would be triggered by switching to a preview tab.

Mr-Russian, 2017-07-22:
If you select Background Processing in general Slic3r settings then every time you make a change, the model will be resliced automatically. Maybe thats what you're looking for?

Sebastianv650, 2017-07-22:
There are reasons why it's switched off by default in Prusa Slic3r now. It makes problems in some circumstates. For example when you edit variable layer heights, it will try to reslice after every mouse click, which means you have to wait some moments after every single change.

Mr-Russian, 2017-07-22:
Oh I see. Makes sense now.

bubnikv, 2017-10-03:
I am closing this one as https://github.com/prusa3d/Slic3r/issues/152 is similar. I referenced this issue from https://github.com/prusa3d/Slic3r/issues/152.

closed 2017-10-03 by bubnikv