prusa3d/PrusaSlicer #438: [FEATURE REQUEST/DISCUSSION] Expose more variables (or even math) to Custom G-Codes

FHeilmann, 2017-07-25:
Title pretty much says it all.

Playing around with stuff like `current_layer` or `bed_temperature` is nice, but being able to use more printer-specific variables would open up a lot of possibilities such as, for example
- printer-agnostic g-codes (e.g. draw a prime line from `0,0` to `0,BED_Y_MAX` 
- more advanced g-code trickeries (e.g. pre-heat to `extruder_temperature*0.5`, probe, heat to `extruder_temperature`)

I haven't really checked the code, but it seems (at least for the exposing more variables part) the facilities are already there.

Hugoz12, 2017-07-25:
I like the idea. I miss the following operation:
- `bed_temperature-5` to preheat the bed, then allow to heat the extruder  to save some startup time
(At the moment I use the other layer bed temperature for preheating but that requieres to have a differences in these two values.)

lordofhyphens, 2017-09-03:
https://github.com/alexrj/Slic3r/issues/3390

https://github.com/alexrj/Slic3r/wiki/Conditional-Gcode-Syntax-Spec

bubnikv, 2017-10-03:
Math parser in the G-code will be useful to satisfy https://github.com/prusa3d/Slic3r/issues/387

bubnikv, 2017-11-19:
At Prusa Research, we are currently trying to merge configs for all our printers into a common base, for which we need to make the linear advance values dependent on the filament AND the printer selected. For that, a conditional filament specific G-code selecting different values for different printer presets would certainly help.

@alexrj and @lordofhyphens proposed a conditional G-code syntax
https://github.com/alexrj/Slic3r/wiki/Conditional-Gcode-Syntax-Spec
based on the discussion in https://github.com/alexrj/Slic3r/issues/3390

@lordofhyphens then later produced a code, 
https://github.com/alexrj/Slic3r/pull/4157
which builds upon https://github.com/ArashPartow/exprtk

We are not going to merge the work of @lordofhyphens, but we are going to implement our own syntax parser for the following reasons:
- Having own parser will allow us to design a language fitting the particular purpose of enhancing a G-code.
- The parser will be aware of the Slic3r configuration types, so it will be able to produce sensible error reports when the values are misused.
- The parser will access the Slic3r configuration store directly for efficiency.

We can build upon the proposal
https://github.com/alexrj/Slic3r/wiki/Conditional-Gcode-Syntax-Spec
but we can diverge as well.

Please provide
- your scenarios for scripting (math, conditional code)
- examples of languages to base our G-code upon
- your concrete proposals of the script syntax

For example, I envision conditional G-code blocks with the syntax loosely based on the C language idioms:
{if condition} some text {elsif condition} some other text {else} yet some text {endif}
{if condition "some text" elsif condition "some other text" else "yet some text}
{switch expression case concrete_value} some text {case other_concrete_value} other text {default}
  default text {endswitch}
{switch expression case concrete_value "some text" case other_concrete_value "other text" default "default text"}

I would support a ternary operator
{expression ? "value when true" : "value when false"}
which would allow me to chain the expression in the sense of the if / elsif above
{expression ? "value when true" : expression2 ? "some othervalue " : "value when false"}

Regular expressions would be nice to have for testing profile names.

Some kind of table selection operator would be nice too. For example, we discussed with Jindra (the guy responsible for the Slic3r profiles) about the syntax for a conditional value dependent on the current printer profile. Something like
M900 K{table printer_name ("MK2 single" => 29), ("MK2 multi" => 200), (default => 5)}
or
M900 K{by printer_name default 35, ("MK2 MM", "MK3 MM") 200, ("MK2", "MK3 pro") 100}
would emit a single value upon the current printer profile name.

The parser will be based on a Boost Spirit library, which will allow me to be pretty flexible with the syntax, but once the syntax is in the wild, it should not be changed, so I am inclined to keep the syntax simple at the beginning and to add complex stuff later after it will be well discussed.

bubnikv, 2017-11-29:
We have a new alpha release implementing basic G-code math and conditional expressions.
https://github.com/prusa3d/Slic3r/releases/tag/version_1.38.2

I am closing this issue. Those of you interested in a richer G-code post processor, please re-open this issue or better create a new one.

closed 2017-11-29 by bubnikv