prusa3d/PrusaSlicer #439: [Research] A level set based method for fixing overhangs in 3D printing

fucx, 2017-07-26:
I just came across this article in "Applied Modelling", a mathematical journal:

http://www.sciencedirect.com/science/article/pii/S0307904X17301002

The suggested method of creating support structures dependant on the model is a bit different from what slicing software usually generates. Due to the nature of these support structures, it might be best to print them with soluble material or work out another method of making them easily removable without leaving scars.

I just wanted to leave this here as a future perspective or discussion starter when it comes to support generation.

4levels, 2017-09-08:
Hi @fucx,

Too bad we don't have access to the article, it seems interesting but you'll need a paying account to read it..

bubnikv, 2017-09-08:
I have found the paper somewhere. If I remember correctly, it applies a
level set method to finding the support volumes. The paper did not seem
convincing to me.

fucx, 2017-09-08:
I have taken a closer look of the PDF version of the article, but I'm not sure I'm allowed to share it. Under no circumstances should one use scihub and look for its DOI number there.

A lot of the math inside the article are way over my head, but the way the supported structures created through their level set method seem promising. I'm neither an expert in 3D printing nor in the required mathematics, but I thought you might find some inspiration there.

closed 2026-09-29 by prusaslicer-bot[bot]