prusa3d/PrusaSlicer #440: Keyboard shortcuts not shown in View menu

freiser77, 2017-07-26:
### Version
1.36.1-prusa3d

### Operating system type + version
macOS Sierra

### Behavior
* In the release notes there is mentioned that the keyboard shortcuts 0-6 are shown in the View menu. But they do not show.

[image]

_Is this a new feature request?_
No, a bug report

bubnikv, 2017-07-26:
Oh well, that's the beauty of multi-platform programming. I used a hack to display an accelerator without being processed by the wxWidgets automatically, and it seems the hack does not work on OSX as I hoped for.

bubnikv, 2017-11-28:
The same issue on Linux: https://github.com/prusa3d/Slic3r/issues/539

bubnikv, 2019-02-03:
The upcoming Slic3r 1.42.0-alpha5 will show the shortcuts, albeit not aligned at the right column. This is currently not possible due to wxWidgets limitations.

closed 2019-02-03 by bubnikv