prusa3d/PrusaSlicer #441: Support for tilted print bed

ahalekelly, 2017-07-26:
There are some new printers available soon, notably the Printrbelt and Blackbelt, which have the build surface at an angle relative to the XY gantry and print head. Printing on these printers now requires a terrible workaround which doesn't work with support material, but it would be much better to have a bed tilt setting built into the slicer. Would this be possible, and what would be the best way to do this?

ManuGithubSteam, 2017-08-06:
AFIK this repo is for the prusa slicer fork only so it will only support the prusa i3 series of printers

ahalekelly, 2017-08-10:
Well this fork obviously works for other printers, and I don't expect them to develop this, I would be doing the work to implement it. But their lack of response indicates how much they're willing to help with a feature that isn't applicable to their printer.

bubnikv, 2017-08-14:
I cannot imagine how the print head would print when the print bed is
tilted. It does not make sense to me.

The nozzle tip has usually a flat part, which has a diameter of roughly
twice the nozzle orifice diameter. The nozzle has to be therefore close to
perpendicular to the surface it squishes the plastic onto, otherwise it
would either dig into the print (and possibly into the bed), or it would
not squish firmly enough.

Do you have any example?

bubnikv, 2017-08-14:
> But their lack of response indicates how much they're willing to help
with a feature that isn't applicable to their printer.

I was on a week vacation, and there are 180 open issues. The feature you
requested is questionable, the whole tilted bed concept developed only to
avoid Makerbot patents. For very most of us, the feature you requested is
of little use.

> Well this fork obviously works for other printers, and I don't expect
them to develop this, I would be doing the work to implement it.

Whether we will be happy or willing to merge your changes depends on the
complexity. It sounds to me that such feature would require large changes
of the code base.

Is there any description of the feature? A good description of the machines
and the software processes required to make use of them?

bubnikv, 2017-10-03:
Sorry, we are not going to implement this feature in a foreseeable future. There is too much to do and this feature is not much promissing.

closed 2017-10-03 by bubnikv