prusa3d/PrusaSlicer #442: excess infill below bridges with 0% infill

bubnikv, 2017-07-26:
### Version
1.36.0, but reproducible with older.

### Behavior

2nd layer below bridges shows the holes to be bridged over.
[image]

1st layer below bridges shows the spurious infills below the bridges.
[image]

Layer with the bridges
[image]

[display-bad-bridging.zip](https://github.com/prusa3d/Slic3r/files/1176636/display-bad-bridging.zip)

tcm0116, 2018-02-05:
This issue is still occurring with 1.38.5 when infill is set to 0%.

[image]

tcm0116, 2018-02-06:
I found a work-around. I created a modifier that encompassed the hole area and the bridges and set the infill density for that modifier to 1 %. It then bridges over the holes as expected and continues using 0 % infill above that point.

bubnikv, 2019-09-06:
Fixed with 48ecbe777f56583723ca8effad7ad01130408599

    Top, bottom and bridging areas are extended into infill as long
    as the infill is not zero. If the infill is zero,
    top, bottom and bridging areas wound not expand into these "void" areas.
    
    With this commit, the top, bottom and bridging areas are allowed to expand
    into the "void" areas as long as these "void" areas are supported below
    with perimeters or some other non-empty infill, and slightly beyond
    these supporting areas into the voids (currently hard coded to 1mm).

The fix will likely be part of PrusaSlicer 2.2.
Closing.

closed 2019-09-06 by bubnikv