prusa3d/PrusaSlicer #443: MM incorrect extruder temperature

jindrichbenes, 2017-07-27:
During multimaterial, Slicer uses extruder temperature of first extruder, not first actually used extruder.
i.e. if extruder 1 is PLA and extruder 2 is ABSm while object starts with extruder 2, printer will preheat heatbed to correct temperature, but uses PLA extruder temperature until first material change.

[mm extruder temperature issue.zip](https://github.com/prusa3d/Slic3r/files/1179214/mm.extruder.temperature.issue.zip)

bubnikv, 2017-07-27:
This only happens if user provided a start G-code with [first_layer_temperature] without defining the initial extruder index. If the user did not provide his M109 code in the start G-code, then Slic3r would substitute its own correctly.

aanon4, 2017-07-31:
The same bug here, but I see this problem if I try to print using a single material, but that material is not the first extruder. The first extruder temperature is selected regardless of the extruder actually being used for printing.

bubnikv, 2017-07-31:
As a workaround, you may replace the section of the start G-code with the following:
M104 S[first_layer_temperature_[initial_tool]] ; set extruder temp
M140 S[first_layer_bed_temperature_[initial_tool]] ; set bed temp
M190 S[first_layer_bed_temperature_[initial_tool]] ; wait for bed temp
M109 S[first_layer_temperature_[initial_tool]] ; wait for extruder temp

bubnikv, 2017-07-31:
Fixed with https://github.com/prusa3d/Slic3r/commit/75c72bc59b3fac35508e5138a64933c124503d64

When first_layer_temperature / first_layer_bed_temperature is used without an index, the value for the active extruder / the initial extruder is used. This applies to all extruder dependent configuration values.

closed 2017-07-31 by bubnikv