prusa3d/PrusaSlicer #444: Start custom GCode inserted too late

pixelk0, 2017-07-27:
### Version
1.36.1

### Operating system type + version
Windows 7 64 bits

### Behavior
Active temperature seting GCODE is inserted before the custom start gcode. I think the custom gcode should be inserted before ANY active (non-comment) GCODE.
You can witness it in the attached GCODE, I also attach my config bundle.

_Is this a new feature request?_ No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

[Config.zip](https://github.com/prusa3d/Slic3r/files/1179784/Config.zip)
[GCode.zip](https://github.com/prusa3d/Slic3r/files/1179785/GCode.zip)

bubnikv, 2017-07-31:
I think this is really a matter of taste. The current behavior is:
1) Set and wait for the bed.
2) Set nozzle temp
3) Call the custom script
4) Call the custom filament dependent script
5) Wait for the nozzle temp.

In my opinion this is reasonable. If you don't like it, you can set the temp using the macros:
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp

Is this not sufficient? What change do you wish and why?

pixelk0, 2017-07-31:
I have a few operations I'd prefer to do with a cold bed/nozzle before the start of the print. So in the current state I'm compelled to reset the heating temp at the start or the custom script. It's both a waste of energy and time.
Also there may be temp presets (PIDs) in the custom start script that could render the temperature set before them ineffective (it's not my case, but you never know). The temperature settings creates a spike in power consumption right at the start of the GCODE, when I have some other power-hungry code to run (turning a LED rail on), which may trip a fuse or PTC protection. And then, but that's a minor point, why call it start script, when it's not run at the start ? 
I see no benefit in stage (1) and (2) being run before the custom script, but maybe I'm missing an important point (I'm what I consider a beginner in those matters).

fiveangle, 2017-07-31:
> I see no benefit in stage (1) and (2) being run before the custom script

It doesn't.  If these are present in start or filament custom gcode:
```
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
```

The auto-set one(s) are not applied.

pixelk0, 2017-07-31:
Ok I didn't understood it disables the pre-start code. It solves my problem, thanks !
There's still a M107 left, but it has little incidence on my prints

bubnikv, 2017-08-01:
M107 only disables the print fan.

So may I close this issue?

pixelk0, 2017-08-01:
It's OK for me, thank you.

closed 2017-08-01 by bubnikv