prusa3d/PrusaSlicer #447: Creation of Perl5 clutter in home dir

Kuro-Maii, 2017-07-28:
### Version

git version: 137aab9

### Operating system type + version

Ubuntu 16.10
$ uname -a
Linux Yggdrasil 4.8.0-59-generic #64-Ubuntu SMP Thu Jun 29 19:38:34 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

### Behavior
* Build.PL creates a folder named Perl5 in the users home folder
* run `build.PL` as the normal user
* no clutter in the users home dir
* appearance of clutter in the form of a Perl5 folder in the users home directory

supermerill, 2017-07-28:
>  ! Can't write to /usr/local/share/perl/5.22.2 and /usr/local/bin: Installing modules to /home/angel/perl5
> ! To turn off this warning, you have to do one of the following:
> !   - run me as a root or with --sudo option (to install to /usr/local/share/perl/5.22.2 and /usr/local/bin)
> !   - Configure local::lib in your existing shell to set PERL_MM_OPT etc.
> !   - Install local::lib by running the following commands

bubnikv, 2017-07-28:
We currently don't have the resources to modify the source code base for
Perl local compilation. I know the mainline Slic3r source has been modified
that way. If you can dig it out, please provide a pull request.

bubnikv, 2017-08-28:
slic3r PE has been updated to store the local modules to a local-lib directory based on the work done at the mainline Slic3r project.

closed 2017-08-28 by bubnikv