prusa3d/PrusaSlicer #448: 1.36.2 still generating missing layers

repetier, 2017-07-29:
### Version
1.36.2 tested 32 and 64 bit windows

### Operating system type + version
Windows probably others as well

### Behavior

Big parts of some models do not get sliced under some conditions. For example
the attached bagclip.
[BagClip_65.stl.zip](https://github.com/prusa3d/Slic3r/files/1184769/BagClip_65.stl.zip)

It is not manifold, but Slic3r 1.3.0 and Slic3rPE 1.36.0 sliced it well. In 1.36.1 I can slice it when I fix the non-manifoldness. But if I combine a fixed and unfixed version both will fail.

So 1.36.2 is better then 1.36.1 but still worse then 1.36.0

#### STL/Config (.ZIP) where problem occurs

bubnikv, 2017-07-31:
Would you please provide more data? config.ini?

I did not see any missing layers on Windows 10 64bit for 1.36.0, 1.36.1, 1.36.2

repetier, 2017-07-31:
Okay it gets now strange. What I did was starting it from repetier host and reimport generated gcode using this command:
C:\Users\littwin\Documents\Projekte\Repetier-Host\src\RepetierHost\bin\Debug\Slic3rPE\slic3r-console.exe --load "slic3r_settings.ini" --dont-arrange -o "composition.gcode" "composition.amf"

on these files:
[SlicingBug.zip](https://github.com/prusa3d/Slic3r/files/1187544/SlicingBug.zip)

And it has only the first 13 layers.

Since you said it worked for you I started the gui, imported the AMF and sliced with same settings and inside your gui it looked in deed right. Then I exported the gcode, imported it to Repetier-Host and it had only 44 layers instead of 50. 6 layers near center were missing.

So I analysed the gcode and the result is that 1.36.2 generates different code that triggers a bug in repetier-host to render the gcode wrong. Here the code

G1 F1200
G1 X89.634 Y92.976 E7.87270
G1 X89.466 Y92.976 F9000.000
G1 F1200
G1 X88.766 Y93.026 E7.87265
G1 X89.426 Y92.953 F9000.000
G1 E7.87270 F1800.00000

I'm not sure if this is a bug in slicer as well. As you see E goes from E7.87270 back to E7.87265 also I'm quite sure it is meant as extrusion. But this tiny retraction without G92 E0 between triggered a bug, which is why I also know that 1.36.0 did not have the retraction at that point. We are speaking here about very tiny values so it might also just be a rounding error from different optimization or a new feature I did not recognize.

So short version having read about the bug in 1.36.1 I assumed it was still there and did not notice it was a double bug on both sides in 1.36.1 and host :-(

So feel free to close issue if you do think the tiny retraction is no real error. In print it will surely not be visible anyway. Will now release new host with 1.36.2.

bubnikv, 2017-07-31:
Would you please do a check for me? If you enable the G-code comments in
Slic3r, how are these buggy G-code lines commented?

repetier, 2017-07-31:
No Problem, here same lines with comments

G1 X90.537 Y93.042 E7.87110 ; infill
G1 F1200
G1 X90.368 Y93.017 E7.87174 ; infill
G1 F1200
G1 X89.955 Y92.985 E7.87265 ; infill
G1 F1200
G1 X89.634 Y92.976 E7.87270 ; infill
G1 X89.466 Y92.976 F9000.000 ; move to first infill point
G1 F1200
G1 X88.766 Y93.026 E7.87265 ; infill
G1 X89.426 Y92.953 F9000.000 ; move to first infill point
G1 E7.87270 F1800.00000 ; unretract
G1 F1200

Seeing all these F1200 slowing communication, how about a filter in output to only set F if it differs from previous value.

Should I wait releasing fixed host?

bubnikv, 2017-07-31:
Thanks for the findings.

I will not release a bugfix build in a near future, so please go ahead and
publish your build.

I will look into the repeated F words one day.

repetier, 2017-07-31:
Ok, I also think it is a small thing.

For curiosity I compared the gcodes - or tried to. But 1.36.0 created much smaller gcode - only 28404 lines instead of 36320 lines. So one reason might be that it did not create the back move is that all moves were a bit longer. But none the less, if firmware decides to retract from such a reduction it would not good, also in this case I guess it is less then 1 step so nothing will happen.

bubnikv, 2018-02-13:
I believe the tiny negative retract should be fixed with https://github.com/prusa3d/Slic3r/commit/ea9920e5d679e2deb19e553ee504cafea7b89714
The fix will be part of 1.39.1-beta.

Regarding the repeated feedrate lines for the infill, we will fix that later due to some difficulties in architecture. Simple solution would be to remove the repeated feedrate lines in a post processor.

bubnikv, 2018-03-19:
The repeated G1 Fxxx entries are now filtered with https://github.com/prusa3d/Slic3r/commit/ebf0ef48f148cbbaa50674492ea0790147ed982d.

closed 2018-03-19 by bubnikv