prusa3d/PrusaSlicer #450: [Windows] Prusa Drivers Version 1.8.0 installer changes settings folder

bagobones, 2017-07-29:
While I am used to the installer overwriting ini files that have the default names, this time it just went and created a brand new folder

C:\Users\Public\Documents\Prusa3D\Slic3r settings MK2 180 instead of C:\Users\Public\Documents\Prusa3D\Slic3r settings MK2 

This seams odd, I am not sure if this is something the installer did or something in the application.. It doesn't really make much sense to move the applications settings unless there was a strong desire to have two versions installed at once.

bubnikv, 2017-11-28:
The installer did it intentionally, so you will not loose your settings.

We will have to come up with some reasonable way of splitting the default presets provided by Prusa Research from the presets defined by the customer. The defaults provided by Prusa Research should become read only, so it would be easy then to replace them by the installer.

bagobones, 2017-11-28:
This will progressively leave a lot of garbage behind the the short term... I would be happy if the Prusa provided configs where static and pushed each time but my custom ones didn't require manual migration every update.

bubnikv, 2017-11-28:
I agree with you it is not optimal and we have plans to improve the configuration layer. We got new people on board, so this issue will hopefully get enough attention.

bubnikv, 2018-02-12:
I am closing this issue, as we are working on updating the profiles from the internet https://github.com/prusa3d/Slic3r/issues/591.

closed 2018-02-12 by bubnikv