prusa3d/PrusaSlicer #451: Linux Appimage icon

fornellas, 2017-07-30:
### Version
1.36.2

### Operating system type + version
Linux Mint

### Behavior
* Appimage has no icon.
* List the Appimage file with Mate's filemanager, caja.
* The Appimage file to have Slic3r icon
* It has no Icon

As an example, Prusa Control's Appimage comes with a nice icon. it'd be nice to see the same for Slic3r.

ManuGithubSteam, 2017-08-06:
Well here on my Arch box i have several Icons for slicer.

Look into /usr/share/slic3r-prusa3d/ there should be some ico files and at least one PNG file  Slic3r_192px.png

You can map these icons to the stsrter menu in the settings i belive...

bubnikv, 2017-12-01:
@probonopd Simon, I am sorry to trouble you again, but do you have an idea what shall I do? By the way, I tested at one college's computer and neither Slic3r PE nor Prusa Control AppImages showed an icon in the file explorer on Gnome on debian (at least I think it was Gnome).

probonopd, 2017-12-01:
Please see https://github.com/AppImage/AppImageKit/issues/346.

bubnikv, 2017-12-05:
@probonopd Do I understand it correctly, that there is no simple solution to implement the request, and that our PrusaControl AppImage cannot show its image by itself as well?

@fornellas Maybe you just assigned an icon to PrusaControl manually and now you mistakenly believe that PrusaControl got the icon assigned by its own?

probonopd, 2017-12-05:
> @probonopd Do I understand it correctly, that there is no simple solution to implement the request, and that our PrusaControl AppImage cannot show its image by itself as well?

True for now, at least until someone (@Azubieta?) implements this functionality for type-2 AppImages.

bubnikv, 2018-02-12:
There is no reasonable way to associate an icon to the AppImage as of now. Closing.

closed 2018-02-12 by bubnikv

probonopd, 2018-02-12:
Well, actually... @azubieta has [implemented](https://github.com/AppImage/AppImageKit/commit/83b82870e1ab03e63178467376be378baca2922e) this in [the optional `appimaged` daemon](https://github.com/AppImage/AppImageKit). So this is how the AppImage looks for me now:

[image]

bubnikv, 2018-02-12:
@probonopd Do I understand it correctly, that now the Slic3r appimage shows the icon already? That's great.

probonopd, 2018-02-12:
Yes, if the user has the optional `appimaged` daemon installed that does automatic integration into the system (it makes AppImages executable, makes them show up in the menus, associates file type associations/MIME types, etc.).

(For MIME types, also see the relevant discussion [here](https://github.com/Ultimaker/Cura/issues/2984).)