prusa3d/PrusaSlicer #452: Slicing error in vase mode

Richie76, 2017-08-01:
### Version
1.34.1-prusa3d-win64

### Operating system type + version
win10pro64

### Behavior
thing:580108
slciing with vase mode
preview shows lines where there shouldn´t be any:
[image]

in the gcode you see some parts where the Z isn´t going upward:
G1 Z22.532 X149.115 Y14.667 E0.01245
G1 Z22.532 X149.004 Y14.859 E0.00750
G1 Z22.533 X148.731 Y15.234 E0.01572
G1 Z22.533 X148.583 Y15.399 E0.00750
G1 Z22.533 X148.249 Y15.704 E0.01529
G1 Z22.533 X148.158 Y15.924 E0.00805
G1 Z22.550 X158.703 Y34.189 E0.71390
G1 Z22.550 X159.035 Y34.036 E0.01236
M204 S1000
G1 Z22.750 F7200.000
G1 X159.009 Y33.981 F7200.000
M204 S800
G1 F2700
G1 X159.856 Y33.591 E0.03157
G1 X160.595 Y33.321 E0.02663
G1 X161.353 Y33.104 E0.02668
G1 X162.124 Y32.940 E0.02668
G1 X162.904 Y32.831 E0.02668
G1 X163.690 Y32.776 E0.02668
G1 X164.477 Y32.776 E0.02663
G1 X165.413 Y32.852 E0.03179
G1 X166.045 Y32.940 E0.02159
G1 X166.816 Y33.104 E0.02668
G1 X167.573 Y33.321 E0.02668
G1 X168.314 Y33.591 E0.02668
G1 X169.034 Y33.912 E0.02668
G1 X169.730 Y34.282 E0.02668
G1 X170.398 Y34.699 E0.02668
G1 X171.035 Y35.162 E0.02663
G1 X171.748 Y35.773 E0.03179
G1 X172.206 Y36.216 E0.02159
G1 X172.734 Y36.802 E0.02668
G1 X173.219 Y37.423 E0.02668
G1 X173.660 Y38.076 E0.02668
G1 X174.054 Y38.759 E0.02668
G1 X174.399 Y39.467 E0.02668
G1 X174.694 Y40.198 E0.02668
G1 X174.938 Y40.948 E0.02668
G1 X175.129 Y41.712 E0.02668
G1 X175.265 Y42.487 E0.02662
G1 X175.382 Y43.785 E0.04411
G1 X196.578 Y43.785 E0.71743
G1 X196.713 Y43.606 E0.00759
G1 X196.804 Y43.197 E0.01418
G1 X196.871 Y42.993 E0.00727
G1 X196.951 Y42.795 E0.00724
G1 X197.157 Y42.407 E0.01486
G1 X197.415 Y42.052 E0.01486
G1 X197.556 Y41.896 E0.00710
G1 X197.885 Y41.599 E0.01500
G1 X198.336 Y41.309 E0.01816
G1 X198.835 Y41.094 E0.01837
G1 X199.262 Y40.988 E0.01491
G1 X199.582 Y40.950 E0.01090
G1 X199.910 Y40.942 E0.01111
G1 X200.348 Y40.988 E0.01491
G1 X200.557 Y41.033 E0.00724
[... 477 lines cut ...]
G1 X147.818 Y15.642 E-0.31364
G1 E-6.28636 F4800.00000
G1 X196.979 Y43.740 F7200.000
G1 E6.60000 F4800.00000
M204 S2000
G1 F2700
G1 X197.240 Y44.001 E0.01264
M204 S1000
G1 Z22.95 F7200.000
M204 S800
G1 F2700
G1 Z22.951 X197.163 Y43.095 E0.02412
G1 Z22.951 X197.293 Y42.797 E0.01103
G1 Z22.951 X197.418 Y42.571 E0.00872
G1 Z22.951 X197.610 Y42.303 E0.01119
G1 Z22.952 X197.779 Y42.109 E0.00869
G1 Z22.952 X198.073 Y41.844 E0.01337

Not exactly sure if this gcode is where the extra lines occur - but the height should be around the same.

_Is this a new feature request?_
NO
#### STL/Config (.ZIP) where problem occurs
since I´m using the prusa slic3r for my cr-10 too this has been modified for that machine.
If you need config let me know.

Sebastianv650, 2017-08-01:
I can't reproduce the issue with my settings, so I think we need your config.

Sebastianv650, 2017-08-01:
You might also try a recent Slic3r release, I just read you are using 1.34.1. It might be fixed in the mean time.

Richie76, 2017-08-01:
Here we go:

[config.zip](https://github.com/prusa3d/Slic3r/files/1191842/config.zip)

bubnikv, 2017-08-02:
I am able to reproduce the issue.

Richie76, 2017-08-21:
Had this happen on another model too.
[image]

this was the model: https://www.thingiverse.com/thing:1535483

If u need to gcode let me know.

Richie76, 2017-09-08:
Hi,

Just curious if there was any progress on this issue.

I came across a model where this issue pops out really significant:
https://www.thingiverse.com/thing:2376777

See the image with hundreds of "infill" where there shouldn´t be any:
[image]

Best regards, Richard

DrLex0, 2017-10-23:
I have noticed the same problems as @Richie76, where infill is added in vase mode. It seems that enabling the ‘ensure vertical shell thickness’ setting is a surefire way to cause the problem as shown in the image from Sept 8, but even with this and all other options disabled that could possibly add extra infill or perimeters, I still get bits of infill scattered across certain vase mode prints. There seems to be some (forgotten?) routine in Slic3r that insists on adding infill in certain situations, and it cannot be disabled. I suspect this same routine might be responsible for [double infill in some cases](https://github.com/alexrj/Slic3r/issues/3829).

In [this attached model + config](https://github.com/prusa3d/Slic3r/files/1407207/vase_config_and_model.zip), little bits of unwanted infill appear and disappear at various places depending on what value I set for infill extrusion width. These interruptions of the spiral are visible as small blemishes on the surface.

[image]
[image]

Vase mode should do its very best to never interrupt the spiral. I would even want to get a warning if it really has to deviate from the spiral once it has been started, or is unable to start it at any point in the model. At this point I'm using Cura for vase mode, because the one in Slic3r is too unreliable.

bubnikv, 2020-02-08:
Spiral vase improvements and bugfixes.
Fixes Connecting / expanding Bottom Layers to Vase Perimeter #253
Fixes Slicing error in vase mode #452
Fixes Slicing Issue (Vase Mode, 0.6mm dmr nozzle) #1887
Fixes Top fill pattern isn't used in spiral vase mode #2533
Fixes Cisar's vase doesn't slice correctly, creates artefacts #3595

When the model is sliced, all the contours are newly oriented counter-clockwise (even holes), merged and then only the largest area contour is retained. In perimeter generator, if the largest contour splits into multiple perimeters, newly only the largest area perimeter is retained in spiral vase mode. These two changes solve #3595 and similar.

The infill is newly calculated only for the bottom solid layers if the spiral vase mode is active (removes various unwanted infill along the vase walls), and the last bottom solid layer is switched to a top solid pattern (solves #2533).

The thin walls are newly enforced to be disabled in spiral vase mode, and the "ensure vertical shell wall" is enforced in spiral vase mode to extend the bottom of the vase to the vase hull (fixes #253).

The changes will be released with PrusaSlicer 2.2.0-alpha4. Closing.

bubnikv, 2020-02-08:
Fixed with 4e11552da980f07216f1720f67f539703ed90ed1

closed 2020-02-08 by bubnikv