prusa3d/PrusaSlicer #453: Some of 1st layer not extending to edge as shown in example file

geodave810, 2017-08-01:
### Version
_Version of Slic3r used goes here_1.36.2

_Use `About->About Slic3r` for release versions_1.36.2-prusa3d-win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_Windows 7 Pro Build 7601: Service Pack 1

### Behavior
* _Describe the problem_Some of 1st layer not extending to edge as shown in example file. STL file was created using openscad.
* _Steps needed to reproduce the problem_Add the enclosed STL file to the slic3r & preview the 1st layer
* _If this is a command-line slicing issue, include the options used_
* _Expected Results_I expected the fan grill pattern to extend to the edge as the STL file was designed
* _Actual Results_Some parts of the grill pattern did not extend to edge on the 1st layer, the rest of the layers look correct
* _Screenshots from __*Slic3r*__ preview are preferred_
[image]

_Is this a new feature request?_No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[Slic3r_TestCase.zip](https://github.com/prusa3d/Slic3r/files/1191653/Slic3r_TestCase.zip)

Sebastianv650, 2017-08-01:
The problem is your 200% default extrusion width in combination with detect thin walls feature enabled. With a that fat extrusion, slic3r seems to find no way how to move to the perimeter lines.

Detect thin walls is a useful feature, but I recommend disabling it in your profiles unless you have a model which realy needs it. Then, carefully inspect the result before printing. Thin walls are usualy not that well attached to normal perimeters and may lead to visible imperfections. It's not the Slic3r doing a bad job, it's just a case of physical limitations.

A wider 1st layer can help with bed adhesion, but you might think a second time about your 200% value. This is a realy hughe extra width..

geodave810, 2017-08-01:
Thanks for that info.  I will turn that detect thin wall feature off.

bubnikv, 2017-08-02:
@Sebastianv650 is mostly right.

1) first_layer_extrusion_width is currently set to 200% of the
first_layer_height. For the first layer height of 0.35, this makes a 0.7mm
width extrusion. This may be a bit high for the 0.5mm dmr nozzle, but it is
not impossible.

2) Disabling the "detect thin walls" feature will solve your issue, with
the price of over-extruding the grill, see the following picture, where the
grill lines are doubled and overlapping by around 70% of their area. The
"detect thin walls" avoids that in most cases.

[image: Inline image 1]

There is a lot of work to be done in regard to the "detect thin walls"
feature to make it robust, and your example is a good test case.

By the way, since Slic3r PE 1.36.1, we have new defaults for the extrusion
widths, if you leave them set to zero. The new defaults are close to the
values we usually set for our printers.

geodave810, 2017-08-02:
Thank you for the clarification.  I have not looked at a lot of those settings in quite a while & usually only change infill pattern or density.  Turning the thin walls off did solve that problem and setting first_layer_extrusion_width to 0 also solved it.  I will leave both of those set to 0 in my standard configuration & adjust them if needed.  I did not see an image in your message, but only a text tag showing an Inline image 1.

closed 2026-09-29 by prusaslicer-bot[bot]