prusa3d/PrusaSlicer #455: how do I upgrade the firmware? I downloaded the zip file, unzipped it, found the file I needed but I cant open a hex file

peterwt, 2017-08-02:
### Version
_Version of Slic3r used goes here_ 1.35.5.16 Prusa3d-win32

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version    windows 7
_What OS are you using, and state any version #s_ 

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2017-08-03:
Which firmware?

If it is MK2, then you will find a firmware uploader utility installed onto
your computer with our Windows drivers.

peterwt, 2017-08-03:
Thanks Guy's. I found the firmware up-loader buried in my computer. Thanks. I'll try to be more precise in asking questions. P

closed 2017-08-03 by peterwt