prusa3d/PrusaSlicer #456: Slic3er generates too few top layers depending on part orientation

hmbusch, 2017-08-04:
### Version
1.36.2 Prusa Edition 64bit version

### Operating system type + version

Windows 10 Home 64bit

### Behavior

I wanted to print the i3 MK2S extruder body and noticed that Slic3r was generating only a single top layer for the part of the model where the v6 heatsink is going to be later, even though I specified 4 top layers. With a 0.2mm layer height, this only top layer is printed at the 3.6mm height mark.

If I rotate the model 180 degrees on the z-axis, the correct amount of top layers (4) is generated at that position, starting at the 3mm mark up to the 3.6mm mark, resulting in 4 top layers.

To reproduce:

* Start Slic3r and add the extruder-body.stl
* Position the part to that the higher portion of it is pointing to the back of the build plate.
* Use the 0.2mm NORMAL profile, adjust top layers to 4 (although the problem also occurs with the preset 6)
* Slice the model
* Only one top layer is generated for the portion that is facing the front of the printer (see screenshots wrong-1.png to wrong-3.png)
* Rotate the model 180degrees along the z-axis
* Slice again
* now 4 top layers are generated (see screenshots correct-1.png to correct-4.png)

#### STL/Config (.ZIP) where problem occurs

[extruder-body-stl-and-config.zip](https://github.com/prusa3d/Slic3r/files/1201416/extruder-body-stl-and-config.zip)

[image]
[image]
[image]

[image]
[image]
[image]
[image]

Hugoz12, 2017-08-04:
Wow weird bug. Can reproduce with 1.35.5.

bubnikv, 2017-11-28:
I have seen a similar issue with one of our objects, and it seems to be caused by numerical issues of the Clipper library that Slic3r uses everywhere. I will have to look into this issue again. The last time I looked into a similar issue I was not able to find a straightforward fix.

bubnikv, 2017-12-14:
There is an issue in the Clipper library when calculating offsets of a contour with holes.

It helps, if you set the Resolution field at Print Settings->Advanced to a nonzero value.

closed 2017-12-14 by bubnikv