prusa3d/PrusaSlicer #458: Implement Cura's Ironing Feature for a more smoother surface finish

m-Th, 2017-08-05:
### Version
Slic3r Prusa Edition
Version: 1.36.2-prusa3d-win64
Build: Slic3r-1.36.2-prusa3d-win64-full-201707281218
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 6.1.7601] 

### Behavior
Please implement a feature which is to be found in the latest Beta of Cura 2.7. In few words, the nozzle travels the top surfaces (pre)heated but without extruding. 
This allows for a smoother finish. 

See it here in action:
https://www.reddit.com/r/3Dprinting/comments/6rkzm7/top_surfaces_using_cura_27s_ironing_feature/

The subthread which speaks about this is here (it is called there neosanding):
https://www.reddit.com/r/3Dprinting/comments/6r35eq/cura_27_beta_is_here/dl28upo/

The general discussion about Cura 2.7 (from which the above subthread is took out):
https://www.reddit.com/r/3Dprinting/comments/6r35eq/cura_27_beta_is_here/

Cura's 2.7 release notes:
https://ultimaker.com/en/blog/50492-introducing-cura-27-beta

_Is this a new feature request?_
yes

Sebastianv650, 2017-08-05:
We had that once back in time: #65 
I think while it seems to work, it's more a nice to have feature than something realy important because it's only aplicable to horizontal top surfaces. As soon as it's tilted by 0.0001°, it's not useful any more. Real world things tend to have non-horizontal planes.

m-Th, 2017-08-05:
1. We had??? (past tense?)
2. Hmmm... depends. Functional parts many times have horizontal surfaces.

Sebastianv650, 2017-08-06:
Non-native speaker ;) Intended meaning: We already had that topic some time ago.

Quintox303, 2017-08-09:
@Sebastianv650 I print for my 3D Hubs customers just about every day, and I'd say that about 90% of the prints I do have a big horizontal surface. I think this would be really useful to implement :)

Quintox303, 2017-08-09:
Please see here https://www.facebook.com/groups/Prusai3users/permalink/795687387282411/ for expressions of interest

fiveangle, 2017-08-09:
With Linear Advance I get top surface quality as good as those "neosanded" pictures :)

I suspect this feature could be a blessing and a curse (I could imagine this completely ruining prints with PETG if not dialed in perfect, plus what about cooking the filament on huge top surfaces ?)

Quintox303, 2017-08-09:
I haven't gotten around to doing the lin advance update yet, I think now's a good time to do it and compare the results :) I'm not sure what the results would be like with PETG, what makes you think it could ruin the print? 

With regards to cooking filament - it does still extrude 10% by default. I'm not sure if that would be enough to stop it cooking though. Personally I've left filament hot in the nozzle for up to half an hour with no "cooking" effects.

Quintox303, 2017-08-09:
Some high resolution comparison pictures below. The bigger part is printed as usual, with no linear advance (haven't updated yet). The smaller disk was printed with ironing in cura. You can see the massive difference - it creates a uniform textured matte surface, which I don't think would be possible to get any other way

[image]
[image]
[image]
[image]

Quintox303, 2017-08-10:
@bubnikv What are your thoughts on this? To me, the surface finish from an ironed print looks very similar to the textured surface that you get from your new powder coated PEI that you use to print the MK2S parts. I think it would be awesome if you could have both the top and bottom surfaces with a similar texture!

daktylus, 2017-08-13:
I get similar smooth top layer finishes just by decreasing the top layer speed down to 5-15 mm/s  for the print.

Quintox303, 2017-08-13:
It's not a smooth surface though. It's a textured surface.. It leaves behind a matte finish which makes the individual extrusion lines impossible to see

bubnikv, 2017-08-14:
I would like to have this feature in Slic3r, but there are currently more
pressing issues, and there is also a lot of work to be done, like a
continuous refactoring to get rid of the unmaintained Perl programming
environment. So this feature is not at the top of my list.

jslick007, 2017-08-14:
I print only functional parts across 2 Prusa MK2s and a MakerBot Rep 2. This feature would be awesome along with separate "designs" for top/bottom layers.

m-Th, 2017-08-14:
> Example, Hilbert Curve for top layers, Rectilinear for bottom.

Definitely. I certainly can use this.

Ok, perhaps it can be done with the https://github.com/prusa3d/Slic3r/issues/378 but there is much more work and more confusing to achieve this.

m-Th, 2017-09-25:
Neosanding 2.0:
https://all3dp.com/neosanding-3d-printing-technique-smooths-your-flat-surfaces/

bubnikv, 2017-09-25:
> your-flat-surfaces/

Maybe you can implement the Neosanding 2.0 in Slic3r just by reducing the
top infill extrusion width 4x? We are going to test it.

m-Th, 2017-09-25:
Make sure to publish this ratio for the user – surely it depends on filament used and situation („more” or „less” neosanding). 

Also, the same stands for (perhaps) some temperature variation (few degrees above / below).

fiveangle, 2017-09-26:
> Maybe you can implement the Neosanding 2.0 in Slic3r just by reducing the top infill extrusion width 4x? We are going to test it.

Neat hack ! :)

m0n5t3r, 2017-12-07:
separate extrusion width for the top layer already exists and has been there for a long time, it's at settings -> advanced -> top solid infill and it does give you that smooth look with certain filaments and speeds

I think the real hurdle to smooth top layers is the path planner now, to be honest; my setup got better since I posted the image in #65, so in Colorfabb's economy PLA it looks like this now (phone slightly misplaced focus and slight issues with linear advance notwithstanding, M3 nut for scale):

[image]

(and yes, it's of rather limited use, since most things don't have flat tops)
(also pinging https://github.com/prusa3d/Slic3r/issues/475#issuecomment-327390981 since it asks for configurable extrusion width)

jettoblack, 2017-12-15:
I’d like to put in another vote for this, neosanding is the only reason I still go back to Cura sometimes. It works great on PLA and PETG too.

kelchm, 2017-12-19:
This would be great to have. If I have some time in the next few weeks I may take a look at implementing this functionality and creating a PR.

supermerill, 2018-01-25:
I made a first draft.
[image]
You need to decouple the top&bottom infill to set only the top one with the smooth feature.
internally, It's a rectilinear then an other rectilinear (flipped 90°) with two time more lines and almost no extrusion.
code : https://github.com/supermerill/Slic3r/tree/smooth_top_infill

rongith, 2018-02-02:
Have an error when exporting gcode from the joined file. The program quit.
[image]
[03033-r06d - I3D-Conduit.zip](https://github.com/prusa3d/Slic3r/files/1689891/03033-r06d.-.I3D-Conduit.zip)
I dismissed the use of brim to the use of mickey mouse pads. It is not important to neosand that pads but the slic3r see that pads as flat surface. It seems the code don't like neosanding first layer.
You could try to solve the bug and the first layer can be neosanded. Or it could be discarded to neosand first layer but that create a new case to manage (new option to tick).

supermerill, 2018-02-03:
@rongith 
try with this:
https://mega.nz/#!81J0lIyK!Vtg2QHjKF4EDRTkfsifQpugb8OCzkutMxn8XN2Dvyoo

i don't have any problem with this version, and no first layer is neosanded.

rongith, 2018-02-04:
@supermerill , managed to export gcode. Thank you. gcode not yet printed .

KDPate, 2018-02-05:
I've been eagerly awaiting this. 
I just did a quick test print, though with no 1st layer areas. It worked great.
It would be nice to have control over the speed (and maybe extrusion rate?) of the smoothing pass.
Thanks for this, it's great!

KDPate, 2018-02-05:
One strange thing: it irons the whole layer if any part of it is the top layer. Or rather it probably irons all solid fill areas, which ends up causing causing some weird behavior depending on the shape of the solid fill (such as a raised ring around the edge of a top surface).

daktylus, 2018-02-05:
I compared it on PLA, top layer speed 40mm/s. It looks nearly identical. I can't even make a proper picture which shows the differene. The neosanded layer loosk slightly more shiny. Could also be the same. Maybe I misused your Slicer. Do I have to activate the neosanding somewhere?

KDPate, 2018-02-05:
Yes, it's not easy to find. Under Infill, you have to change the top fill pattern to "smooth".

daktylus, 2018-02-06:
It worked. Printed in PLA (dasFilament), with Linear Advance, top layer speed: 40 mm/s

**Left two pieces normal, right one with neosanding.**

[image]
[image]
[image]

bubnikv, 2018-02-13:
@supermerill Thanks for your great effort. Would you please create a pull request?

There is one thing I believe you did not do, and which makes sense to add: Slic3r extends the top surfaces inwards into an object by a fixed amount. By the way, the extension of the surfaces inwards aka anchoring of the full infill surfaces is the reason of https://github.com/prusa3d/Slic3r/issues/668. It really makes sense to complicate things and to do the neosanding over the visible surface only, not over the extended surface.

supermerill, 2018-02-13:
@bubnikv 
My changes are made vs a 18/1/4 commit.
Before making the pull request, I move my change to the current master or an other commit?
Do you prefer one pull request by modification or a pull request for the whole package?
(gyroid infill, ironing, only 1 perimeter top)

For the Visible thing, it's possible. All my modification use is_external() && surface_type == stTop.

daktylus, 2018-02-13:
@supermerill  
Can I make some wishes for the implementation? I think the feature is quite hard to find and "smooth" doesn't describe the routine properly (also using 10 mm/s on top layer gives you a SMOOTH finish). I think a seperate checkbox with some label like: "Use neosanding on top layer" or "hypersmooth top layer" would be much clearer.

supermerill, 2018-02-13:
@ daktylus the other names are short. I can put "neosanding" or "ironing".

KDPate, 2018-02-13:
Yay, ironing!
Speed control for the ironing pass?

supermerill, 2018-02-13:
Maybe in v2, when the gui conversion will be over.

apphead, 2018-07-21:
Cannot find top fill pattern "smooth". Is iroing still a dev feature or just hidden somewhere I didn't look?

supermerill, 2018-07-21:
it's not merged in the main branch, you have to get my fork (I release a win32 build in the pull request thread) to use it.

fdev31, 2018-09-25:
It is partially covered by the "layer editing" option... it's not exactly the same but if you add one vertical layer + ensure the top ones are very thin you may achieve a comparable result.

rongith, 2018-09-25:
@fdev31. I disagree and your comment should probably be base on your specific usage. I wish Prusa add ironing in its version. @supermerill done a very good job.
Per example, you may have many top layers then you can't simply change layer height manually. Final result without ironing is sensible to material. I use a lot a 0.30 layer height where ironing is very effective.

fdev31, 2018-09-26:
Actually this is a way I found to work around it, it's not "replacing" the feature, but still allows very nice top layers... I just change the height of those layers. I completely agree that it depends on the use case and it's some kind of tedious workaround for many models.
A one click option doing just the right thing is definitely something I would like to see in slic3r too... I just wanted to drop a tip in case it helped someone... with very thin top-most layers I get excellent results too, and it works on non-flat surfaces as well...

probonopd, 2018-09-27:
> I wish Prusa add ironing in its version.

:+1:

rongith, 2018-10-01:
@fdev31, I was afraid you dismissed the usefulness of that feature. I will try to remember your tip if it simply solves a case and supermerill release doesn't work anymore.

probonopd, 2018-11-19:
This is an example of a part that could benefit from the feature:

[image]

Murloc992, 2018-12-10:
Any news on this? I really loved this feature..

KyleMaas, 2019-02-02:
I would love to see this feature implemented as well.

3dprintingrookie, 2019-02-11:
I'm waiting for this feature, please add it!

cyxlops, 2019-02-15:
When can we get this!? Why won’t the slic3r developers reply and are completely ignoring this topic?

Quintox303, 2019-02-15:
This is already a part of supermerril's branch, it just needs to be ported across

cyxlops, 2019-02-15:
When will they port it? Supermerril had this feature for a while and they didn’t bother porting it. It would be a nice gesture for them to acknowledge his hard work and implement his feature. It’s only fair.

vojtechkral, 2019-02-15:
@cyxlops it's not a matter of bad intent or ingratitude or something, it's just that no one yet had the cycles to do it...

bubnikv, 2019-02-17:
> When will they port it? Supermerril had this feature for a while and they
didn’t bother porting it. It would be a nice gesture for them to
acknowledge his hard work and implement his feature. It’s only fair.

It is more complex than that. First, we did not receive an implementation
by @supermerill, that was provably useful, at least by our measures.
Merging something that is not provably better than no ironing does not make
sense from maintenance point of view.

Second, we need to invest time of some of our engineer to review the code
changes. Imagine "merging" some changes that somebody claims to do only
good to your car, into your engine, without reviewing all the effects this
change will possibly have. I wish you a good luck. We cannot do that. We
are a business, and we are responsible to deliver a stable product to our
customers.

This is not a complaint to @supermerill 's great work. I wish we would have
more time to cooperate.

3dprintingrookie, 2019-02-18:
@vojtechkral  @bubnikv Yes you are a business and I understand how complex it is create a stable program and I really appreciate all the effort and work especially with the blog updates @josefprusa keeps emailing us. It's just that the ironing feature has been requested for a while and we would love to see it! Also, I'm certain that @supermerill would cooperate and explain his code and specifically the ironing feature to help you migrate it into your core.

If it's difficult to migrate a code from another user, why not design it from scratch? It's been requested since 2017, would you not think that some dedicated time might be given to slowly implement this feature since 2 years ago? I kinda agree with @cyxlops and the other people here (and reddit) who want this feature implemented, it really enhances how 3D objects turn out. When Cura released this feature, so many people went online to submit pictures of their prints with better surfaces, and in this way advertising Cura further. This is how I got to know about Cura in the first place tbh.

Again, I appreciate all the work you guys are putting into Slic3r, it's miles better than what it was since release.

3dfanantic, 2019-02-21:
I wanted to post this but I noticed it is already posted so I came here to comment that pls add ironing! I dont like cura and want to use slic3r only but ironing is very good feature

aderusha, 2019-02-25:
Another vote here! This is something that makes an immediate and apparent difference on print quality, particularly for those of us creating functional prints (whomever said most models don't have flat top surfaces must be working with figurines I suspect).

mperino, 2019-02-26:
Another vote here, Supermerril's branch doesn't have a Linux Appimage, and doesn't compile for me using the standard Prusa SlicerPE Linux compile instructions.  Really would LOVE to have this as a solution for parts that I print with a small nozzle, at 0.5 layers.  It just feels so wrong to slic3r in a windows VM just for this feature :(

mperino, 2019-02-26:
And most importantly of all THANK YOU @vojtechkral @bubnikv and  @supermerill  Really appreciate all your hard work!

Rycoviac, 2019-03-06:
Yep, I'm on this feature-request, too. This finishing-option should be in Slic3r PE. I'm short of changing to Cura because of this, since it creates a way more better finish for customers and also closes small gaps.

OopAck, 2019-03-07:
Please add Ironing top and bottom layers to Slic3r. 
This version has it.
https://github.com/supermerill/Slic3r/releases?fbclid=IwAR3tUem6pK3ifoNSK2hERF_zyiVC9znRaHB8Oj-SpfjBZd2f71i0CYyXQk0

mhavo, 2019-03-12:
One vote more.

[81 more replies]

closed 2020-09-27 by lukasmatena