prusa3d/PrusaSlicer #460: layer height setting for parts of object

mhajicek, 2017-08-07:
### Version
1.36.2

### Operating system type + version
Win7 Pro 64bit

### Behavior
When "layer editing" button in "Plater" tab is used for an object, then it blocks usage of "settings button" -> Layers. But it stays unusable (entered layer heights are ignored) even if all changes in layer height done by usage of "layer editing" are reset to what it was before, so there is no conflict anymore.

Both approaches have very good usecases though so both should be preserved - layer editing approach beeing better for curved or art stuff, direct input of exact layer height for some layer ranges beeing better for technical components, depending on shape.
Would be great to connect these two features: layer editing should put changes made by mouse to the layers height list where it could be later viewed or corrected, and vice versa, changes made by directly writing layer heights to the list should be visible in the layer editing vertical bar.

_Is this a new feature request?_
yes

m-Th, 2017-08-08:
IOW: Both should be editors for the same internal data structure.

bubnikv, 2017-12-05:
The problem is, you would get too many entries in the table if you went from the graphical editor to the table. Therefore I am keeping them separate.

You may start with the table and then edit the table with the graphical editor, but not vice versa.

rtyr, 2020-03-22:
I believe this is not a problem anymore with current PrusaSlicer. Closing.

closed 2020-03-22 by rtyr