prusa3d/PrusaSlicer #461: Z Seam commanded underextrusion/nonextrusion/coast error at end of layer moves

ReflectorL, 2017-08-08:
### Version
1.36.0

### Operating system type + version
Win7/64, run inside of Sandboxie.

### Behavior
* _Describe the problem_
When models have vertical walls of 0.8mm thickness with a 0.4mm nozzle, the gcode seems to call for the printer to do some kind of reduced extrusion rate (Like as if it was trying to coast) towards the end of a layer, consistently in the same spot which causes a weakened wall. This behavior does not appear to manifest when wall thickness is 1.2mm. It does not appear to be an underextrusion problem as the rest of the print will be fine otherwise. This consistently creates weak walls with "pitting" that is actually caused by an extrusion that appears to be like a thin wisp followed by a blob and thin wisp, repeating. It appears to occur even with layer thickness set to .15 instead of .3. It consistently occurs at the same part of a model. On tall models that have vertical walls, this appears to occasionally change parts on the model instead of propagating straight up (Ex: From the "front right" of a model to the "rear left") every so many layers (Measurably >1-2cm sections, but less than say, 5cm), as long as seam alignment is set to nearest.

* _Steps needed to reproduce the problem_
Seam set to nearest

* _Expected Results_
[image]

* _Actual Results_
See right
[image]

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
[config.zip](https://github.com/prusa3d/Slic3r/files/1207711/config.zip)
[Battery Case Bottom.zip](https://github.com/prusa3d/Slic3r/files/1207713/Battery.Case.Bottom.zip)

bubnikv, 2017-08-14:
What 3D printer is it printed on? What material? Is it PET?

It seems to me like your printer / print material does not like the rapid
changes in extrusion rate caused by the gap fill.

ReflectorL, 2017-08-19:
Wanhao Duplicator 6 with an enclosure. Material was PLA @ 240C set at the printer.

This does not seem to occur with random instead of nearest for the seam position.

bubnikv, 2017-10-03:
It may be a similar issue to https://github.com/prusa3d/Slic3r/issues/489

It seems to me to be a property of your Wanhao. Maybe the machine has a rough or pointy nozzle? Maybe there is insufficient print cooling? We certainly do not see this issue with our Prusa MK2 / MK3 printers.

closed 2026-09-29 by prusaslicer-bot[bot]