prusa3d/PrusaSlicer #463: exponentional values generated as G1 F-1.95058e+006 causing problems #382

gruvin, 2017-08-09:
### Version
1.36.2-prusa3d

### Operating system type + version

macOS 10.12.6

### Behavior

As per previously closed issue #382. 

Exported gcode in the attached ZIP file contains lots of negative F values, as in line 1373 ...

```
G1 F-136105
```

... and many more.

#### STL/Config (.ZIP) where problem occurs
[Bad-GCODE-Slic3r-v1.36.2-prusa3d_macos.zip](https://github.com/prusa3d/Slic3r/files/1210307/Bad-GCODE-Slic3r-v1.36.2-prusa3d_macos.zip)

gruvin, 2017-08-09:
Ah! Changing `Max volumetric speed` from 8 to 0 appears to resolve the problem. I only just set that today, in this version at least.

bubnikv, 2017-09-12:
The following commit should fix it: https://github.com/prusa3d/Slic3r/commit/6b2b2798890e34eff7ff148c7c38b6b90988472b

It was a regression bug. Thanks for the report. Closing.

closed 2017-09-12 by bubnikv