prusa3d/PrusaSlicer #465: Adaptive (Slope sensitive) infill / supports

m-Th, 2017-08-10:
### Version
Slic3r Prusa Edition
Version: 1.36.2-prusa3d-stable-win64
Build: Slic3r-1.36.2-prusa3d-stable-win64-full-201707281104
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 6.1.7601] 

### Behavior
Implement an algorithm which will take in account the slope (and, perhaps, other factors) in order to optimize the infill (when the infill is used as an internal support) and, more importantly, the supports generation in order to optimize the print time and used material.

This feature is present in other slicers on the market.

_Is this a new feature request?_
yes

bubnikv, 2017-10-03:
>Implement an algorithm which will take in account the slope (and, perhaps, other factors) in order to optimize the infill (when the infill is used as an internal support) and, more importantly, the supports generation in order to optimize the print time and used material.

You mean an adaptive infill and support with increasing density when reaching the overhangs?

> This feature is present in other slicers on the market.

Which slicers? I am aware of the Cura's adaptive cubic infill.

m-Th, 2017-10-04:
> You mean an adaptive infill and support with increasing density when reaching the overhangs?

Yes

> Which slicers? I am aware of the Cura's adaptive cubic infill.

Yes. :)

Beside Cura we have Autodesk Meshmixer with a very interesting (and customizable) solution:

[image]
(with color: supports; grayscale: the actual object)

You can download it (it is free) and play with the options.

bubnikv, 2018-03-14:
For reference, Cura implementation of the adaptive cubic infill: https://github.com/Ultimaker/CuraEngine/issues/381

bubnikv, 2020-11-20:
Adaptive cubic infill has been implemented in PrusaSlicer 2.3.0-alpha3.
Closing.

closed 2020-11-20 by bubnikv