prusa3d/PrusaSlicer #466: Adaptive (Slope sensitive) automatic layer height & other improvements

m-Th, 2017-08-10:
### Version
Slic3r Prusa Edition
Version: 1.36.2-prusa3d-stable-win64
Build: Slic3r-1.36.2-prusa3d-stable-win64-full-201707281104
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 6.1.7601]

### Behavior
There is a lot of discussion today about automatic layer height generation: 
https://www.reddit.com/r/3Dprinting/comments/6rvur9/im_in_the_process_of_making_an_automatic_variable/

...however a thing which seems straightforward can have undesired secondary effects, that's why the corresponding code in Slic3r PE is disabled (!). Relevant comment is here:

https://www.reddit.com/r/3Dprinting/comments/6rvur9/im_in_the_process_of_making_an_automatic_variable/dl8erhr/

That's why I propose to add an "Auto" button above the "Reset" button on the Layer Editing pane (in the right part of the 3D viewer with the model).

By pressing the "Auto" button a dialog should appear asking:

_Do you want to generate the automatic layer heights for the edited area only?
Press [Yes] to apply to the edited area, [No] to apply to the entire object or press [Cancel] to abort the operation.
[Yes], [No], [Cancel]_

I think that the dialog is self explanatory:
When the user presses Yes (the default option), the algorithm will be applied only on the selected area.
When the user presses No, the algorithm will be applied to the entire object, but the user will have the possibility to edit/reset (parts of) the algorithm's result.

_Is this a new feature request?_
yes

doominabox1, 2017-08-12:
I agree with this post ;)

m-Th, 2017-08-14:
Here's an analysis on this topic (Adaptive Slicing):
http://staff.ustc.edu.cn/~lgliu/Projects/2015_AdaptiveSlicing_3DPrinting/Papers/2015_CGF_AdaptiveSlicing.pdf

bubnikv, 2017-08-15:
> Here's an analysis on this topic (Adaptive Slicing):
>http://staff.ustc.edu.cn/~lgliu/Projects/2015_AdaptiveSlicing_3DPrinting/Papers/2015_CGF_AdaptiveSlicing.pdf

Thanks for the hint, this sounds cool. I am coming from a computer graphics community, but I was not aware of this paper.

My approach to the adaptive layer height would be a kind of surface sampling using random rays thrown from the infinity. A layer would only be thinned, if it has low angle surfaces and those surfaces collect enough samples, meaning they are visible from the outside. After this step, the profile would have to be filtered to be smooth.

m-Th, 2017-08-15:
> My approach to the adaptive layer height would be a kind of surface sampling using random rays thrown from the infinity. A layer would only be thinned, if it has low angle surfaces and those [...]

This sounds cool too. :)

I think that it is a reasonable approach which can be effective.

Be sure tough to allow the user to chose the area (entire object / edited area) on which this will be applied. There are some things (for example letters, functional features like handlers etc) which we want to be with thick layers in order to gain speed.

gregspruce, 2017-11-29:
I've recently started using KISSlicer about half of the time due to the better wipe tower implementation by PJR and the adaptive layer heights (which also allow below .15 for multi-material without any issue by the way). I would love to get a similar option for auto adaptive layer height in Slic3r.

dhxgit, 2019-02-18:
I would be curious if this has a chance of being implemented in future?

I think that messing around manually with the layer height is unnecessary if it can be done automatically according to the slopes (which is most likely that what you do manually anyway).

SvetoBobr, 2019-10-20:
Don't care about the dialogs but adaptive slicing like in the original slic3r is crucial for those who work in command line and/or use scripts to process models in bulk.

lukasmatena, 2023-02-22:
Automatic variable layer height was implemented years ago. Closing.

closed 2023-02-22 by lukasmatena