prusa3d/PrusaSlicer #468: Build gui fail Wx

SumFungi, 2017-08-11:
### Version
version_1.37.0-7-g3b54b68b

### Operating system type + version
Win10 v.1703

### Behavior
command line builds and prompted to build gui but it had error:
--> Working on App::cpanminus
Fetching http://www.cpan.org/authors/id/M/MI/MIYAGAWA/App-cpanminus-1.7043.tar.gz ... OK
Configuring App-cpanminus-1.7043 ... OK
Building and testing App-cpanminus-1.7043 ... FAIL
! Installing App::cpanminus failed. See C:\wperl64d\cpanm\work\1502484802.6972\build.log for details. Retry with --force to force install it.

When attempting to build GUI with:
perl Build.PL --gui

Found Wx 0.9932 which doesn't satisfy 3.0.
The following prerequisites failed to install: Wx

I'm using all pre-built libraries and files from wiki links.
https://github.com/prusa3d/Slic3r/wiki/How-to-compile-Slic3r-Prusa-Edition-on-MS-Windows

bubnikv, 2017-08-14:
> Building and testing App-cpanminus-1.7043 ... FAIL
> ! Installing App::cpanminus failed. See C:\wperl64d\cpanm\work\1502484802.6972\build.log for details. > Retry with --force to force install it.

What does C:\wperl64d\cpanm\work\1502484802.6972\build.log say?

SumFungi, 2017-08-14:
cpanm (App::cpanminus) 1.7042 on perl 5.024000 built for MSWin32-x64-multi-thread
Work directory is c:\wperl64d\cpanm/work/1502484802.6972
You have make "C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\BIN\amd64\nmake.exe"
You have LWP 6.15
Falling back to Archive::Tar 2.04
Searching App::cpanminus () on cpanmetadb ...
--> Working on App::cpanminus
Fetching http://www.cpan.org/authors/id/M/MI/MIYAGAWA/App-cpanminus-1.7043.tar.gz
-> OK
Unpacking App-cpanminus-1.7043.tar.gz
Entering App-cpanminus-1.7043
Checking configure dependencies from META.json
Checking if you have ExtUtils::MakeMaker 6.58 ... Yes (7.24)
Configuring App-cpanminus-1.7043
Running Makefile.PL
Checking if your kit is complete...
Looks good
Generating a nmake-style Makefile
Writing Makefile for App::cpanminus
Writing MYMETA.yml and MYMETA.json
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have Test::More 0 ... Yes (1.302078)
Checking if you have ExtUtils::MakeMaker 0 ... Yes (7.24)
Building and testing App-cpanminus-1.7043

Microsoft (R) Program Maintenance Utility Version 12.00.21005.1
Copyright (C) Microsoft Corporation.  All rights reserved.

cp lib/App/cpanminus.pm blib\lib\App\cpanminus.pm
cp lib/App/cpanminus/fatscript.pm blib\lib\App\cpanminus\fatscript.pm
	"C:\wperl64d\bin\perl.exe" -MExtUtils::Command -e cp -- bin/cpanm blib\script\cpanm
	pl2bat.bat blib\script\cpanm

Microsoft (R) Program Maintenance Utility Version 12.00.21005.1
Copyright (C) Microsoft Corporation.  All rights reserved.

	"C:\wperl64d\bin\perl.exe" "-MExtUtils::Command::MM" "-MTest::Harness" "-e" "undef *Test::Harness::Switches; test_harness(0, 'blib\lib', 'blib\arch')" t\*.t
# App::cpanminus/1.7043
t\happy_cpantesters.t .. ok
All tests successful.
Files=1, Tests=1,  1 wallclock secs ( 0.08 usr +  0.00 sys =  0.08 CPU)
Result: PASS

Microsoft (R) Program Maintenance Utility Version 12.00.21005.1
Copyright (C) Microsoft Corporation.  All rights reserved.

WARNING: Unable to unlink 'C:\wperl64d\bin\cpanm': Permission denied
Going to try to rename it to 'C:\wperl64d\bin\cpanm.AAA'.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
WARNING: Rename failed: Permission denied. Scheduling 'C:\wperl64d\bin\cpanm.AAA'
for installation as 'C:\wperl64d\bin\cpanm' at reboot.

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
ERROR: MoveFileEx 'C:\wperl64d\bin\cpanm.AAA' for installation as 'C:\wperl64d\bin\cpanm' at reboot failed: Access is denied
You may try to perform the operation by hand yourself. 
(You may need to close other perl processes first).
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
 at -e line 1.
Installing C:\wperl64d\site\lib\App\cpanminus.pm
Installing C:\wperl64d\site\lib\App\cpanminus\fatscript.pm
NMAKE : fatal error U1077: '"C:\wperl64d\bin\perl.exe"' : return code '0x2'
Stop.
-> FAIL Installing App::cpanminus failed. See C:\wperl64d\cpanm\work\1502484802.6972\build.log for details. Retry with --force to force install it.

bubnikv, 2017-08-14:
try to make all files under C:\wperl64d writable.

SumFungi, 2017-08-16:
I updated my permission and applied the setting to make everything in C:\wperl64d writeable. It looks correct. Doing that doesn't seem to resolve the issue with but reloading the CMD shows cpanimus as up to date with 1.7043. The command line builds successfully but the gui build still fails with the WX error:

Found Wx 0.9932 which doesn't satisfy 3.0.
The following prerequisites failed to install: Wx

I noticed wxperl is a different version number, 0.9931 than wxwidgets, 3.0.3

Geartechbrandon, 2017-08-22:
I was able to recreate your issue in a Win10 VM, still not sure why, but you might want to consider building Perl and Wx from source for your system. You can follow the instruction on the [Wiki link](https://github.com/prusa3d/Slic3r/wiki/Building-Perl-&-wxWidgets-with-Visual-Studio-2013) to do this. 

If it's just for your own personal use it's good to know that even when WX fails to be "up to date" you can still run slic3r with:
`perl slic3r.pl`

Follow through the setup wizard and the program will error and exit. Execute
`perl slic3r.pl`
again and everything should function normally.

bubnikv, 2017-08-28:
> Found Wx 0.9932 which doesn't satisfy 3.0.
> The following prerequisites failed to install: Wx
> 
> I noticed wxperl is a different version number, 0.9931 than wxwidgets, 3.0.3

That seems to be a new bug on my side. You may try to build the latest released source code.

bubnikv, 2017-08-28:
I have fixed the minimum Wx version with 
https://github.com/prusa3d/Slic3r/commit/cf2f29a5f11a13e436e8a10ed04894818f438270

bubnikv, 2017-09-12:
@SumFungi Does it work for you now?

bubnikv, 2017-10-03:
no response, closing.

closed 2017-10-03 by bubnikv