prusa3d/PrusaSlicer #469: First layer skirt only has half the loops when using multiple extruders

deamon002, 2017-08-11:
### Version
Version: 1.36.2-prusa3d-win64, been able to reproduce starting from 1.35.1.

### Operating system type + version
Windows 10 Pro 64

### Behavior
* _Describe the problem_
When slicing an object with a skirt, if more than one extruder is used in Print Settings -> Multiple Extruders -> Extruders, the actual G-code generated will only produce half the number of loops for the skirt on the first layer (rounded up). Any skirt printed on subsequent layers (like for ooze prevention) has some of the loops printed in mid-air. 
This only shows up in the generated G-code, the preview does not show it.
* _Steps needed to reproduce the problem_
1. Add an object.
2. Set minimum number of skirt loops to a value higher than 1.
3. Set a different extruder for support material/raft/skirt.
4. Export G-code.
5. Open the file in a G-code viewer. The first layer skirt will only have half as many loops as set in the Print Settings. Any layers above the first will have the right number (if skirt height is set to more than one layer or using ooze prevention).
* _Expected Results_
Same number of skirt loops on all layers.
* _Actual Results_
Only half the number of skirt loops on the first layer (rounded up).
* _Screenshots from __*Slic3r*__ preview are preferred_
Issue only shows up in generated G-code, not in the preview.
* _Is this a new feature request?_
No 

#### STL/Config (.ZIP) where problem occurs
[slic3r_skirt_bug.zip](https://github.com/prusa3d/Slic3r/files/1219182/slic3r_skirt_bug.zip)

bubnikv, 2017-08-14:
Thanks, I see the issue.

The Slic3r PE hast the G-code rewritten from Perl to C++ and reworked to minimize the number of tool changes, but the basic logic for the skirt is by @alexrj:
The 1st layer skirt is extruded with possibly all extruders possibly printing. In your case, you have the supports enabled, so it would prime the support nozzle as well, if there was a support at the 1st layer.
The other layers will have the skirt printed with the active extruder completely, so the other nozzles will not be primed.

What are you using the skirt for? Shouldn't Slic3r prime every nozzle over the skirt after each tool change on any layer?

deamon002, 2017-08-14:
Do you think that tool change minimization is the cause? Since it only seems to be priming the one extruder, is it skipping the ones that aren't active, but still generating the loops for the active one as if the others won't be skipped? I did notice that if you set perimeter and infill to one extruder, support/raft/skirt to another, and interface to a third extruder, then the number of first layer skirt loops is cut in threes instead of half.

I was trying to find the settings to get support material to work. I had it constructing a 3-layer raft, to test the adhesion of the build material to the support material. It didn't stick very well, so I was looking for possible causes, and noticed quite a lot of material was oozing out of the inactive extruder, so I turned on ooze prevention. Unfortunately, the tall skirt that builds around the model wouldn't stick, and I just ended up with threads of material being extruded in mid-air, being dragged all over the place and messing things up. After a while I noticed the base layer of the skirt was only partially being printed, so I went from there trying to isolate which settings cause that behavior.

bubnikv, 2017-08-15:
> Do you think that tool change minimization is the cause?

It is the reason of this bug. It is certainly a bug, I will fix it.

My question is: How shall the multi-loop skirts be printed with multi-material prints? Shall the loops of the skirt be used to prime all the extruders active on the particular layer? Or shall the skirt be printed just with the 1st active extruder of the layer?

gregspruce, 2017-08-15:
>> Do you think that tool change minimization is the cause?
>It is the reason of this bug. It is certainly a bug, I will fix it.

>My question is: How shall the multi-loop skirts be printed with multi-material prints? Shall the loops of >the skirt be used to prime all the extruders active on the particular layer? Or shall the skirt be printed >just with the 1st active extruder of the layer?

I would love if each extruder in a layer would do it's portion of the skirt before it begins it's portion of the layer.

My preferable logic would be:
-1st active extruder for layer does 1 complete skirt loop
-1st active extruder does its portion of the layer
-Print head moves out of skirt to do the temp delta for tool change (for multi-nozzle printers, would also 
   love a software option to set nozzle offset for this feature to make sure all nozzles are outside of skirt 
  that doesn't effect the actual print since my nozzle offset is set in firmware)
-2nd extruder does some portion of the skirt to ensure fully primed (complete loop not necessarily 
  required for oozing since 1st extruder did this
- 2nd extruder does it's portion of the layer
...rinse and repeat

bubnikv, 2020-01-14:
The skirt generator used to prime all printing extruders at the 1st layer if enough skirt lines were configured, while at the other layers the skirt used to be printed with the 1st extruder printing the layer.

There was a bug introduced quite a long time ago, where not enough skirt lines were extruded if some extruders were not needed to print the 1st layer.

Newly the skirt generator works the same way on all layers: All the extruders planned to print a layer are primed over the skirt if enough skirt lines are configured.

Fixed with 10d1b45. The fix will be part of the next PrusaSlicer 2.2.0 alpha. Closing.

closed 2020-01-14 by bubnikv