prusa3d/PrusaSlicer #470: Enforce support for first N layers has no effect

DrLex0, 2017-08-12:
### Version
1.36.2

### Operating system type + version
Mac OS X 10.11.6

### Behavior
* The ‘Enforce support’ feature does not work.
* _Steps needed to reproduce the problem_
  * Slice the attached model with attached settings.
* _Expected Results_
  * It looks somewhat like this (screenshot from a recent dev build of the ‘vanilla’ Slic3r): [image]
* _Actual Results_
  * It looks like this: there is no support to be found. [image]

While I'm at it (who knows, maybe related), a small nuisance in the UI: toggling the “Generate support material” checkbox is impossible as long as either the “Overhang threshold” or “Enforce support for the first” text fields have focus.

#### STL/Config (.ZIP) where problem occurs
[config_and_model.zip](https://github.com/prusa3d/Slic3r/files/1219844/config_and_model.zip)

DrLex0, 2017-08-12:
Never mind, is duplicate of #273

DrLex0, 2017-08-12:
… but, that fix should be in 1.36.2, so reopen

bubnikv, 2017-08-14:
Did you have the supports enabled? In your config the supports are off.

DrLex0, 2017-08-14:
The tooltip says: “regardless of whether normal support material is enabled or not”.

bubnikv, 2017-08-14:
I see, this is actually new to me. I will have to think over it, whether to maintain the 
“regardless of whether normal support material is enabled or not”
or not. Frankly I was not aware of this exceptional case.

DrLex0, 2017-08-14:
I'm not sure if this was recently changed, but I seem to remember from older versions that supports as a whole needed to be enabled for this to work, so I found the tooltip a bit surprising as well. I'm not sure what the desired behavior should be either. It does make some sense to keep it independent from the normal supports. With the current behavior, if one only wants these bottom layer supports, the overhang threshold must be set to 1° to prevent generating supports higher up, which is a bit of a hack…

Sebastianv650, 2017-08-14:
@bubnikv, we had that once already: #318

bubnikv, 2017-08-14:
@Sebastianv650 Sorry, I did not read #318 carefully enough.

3axis, 2017-09-01:
I am finding supports difficult also, something I sliced in 1.36.2, looked like it was supported but when it printed the thin bottom edge was slightly in the air resulting in mis-shapen edge and a junk print. I tried adjusting many options including enforce for first 25 layers. Nothing made any difference really, could not get a successful print. I know the model is awkward but it is now printing looking ok from Cura. I love Slic3r and the Prusa layer heights, it would be fantastic if supports got hugely improved.

[image]

[DehydratorBracket.stl.zip](https://github.com/prusa3d/Slic3r/files/1269867/DehydratorBracket.stl.zip)

I put a lengthy comment on this article because I would love to see Slic3r supports improve, particularly the possibility to create manual supports: http://www.prusaprinters.org/printing-soluble-interface-supports-prusa-i3-mk2-multi-material/

boelle, 2017-09-19:
i had the same problem with the attached model

support was only generated under the "tubes" and all the way up in the tubes

i put it on the plater with the tubes up as it would require the least amount of support

the print came out very badly as the slicer created overhang from the 2 small sides... ie the biggest distance possible

[image]

[image]

i run slic3r from within repetier host as i need the gcode to be repetier firmware specific

i have tried to slice direct from slic3r (latest version) and its the same there

[Huller_original.zip](https://github.com/prusa3d/Slic3r/files/1314066/Huller_original.zip)

bubnikv, 2017-11-28:
Another report of the same issue: https://github.com/prusa3d/Slic3r/issues/541

bubnikv, 2021-02-23:
Fixed with d2c4602

closed 2021-02-23 by bubnikv