prusa3d/PrusaSlicer #472: Appimage incompatable with expat > 2.2.1

jkoelker, 2017-08-13:
### Version
1.36.2 (really any of them ;)

### Operating system type + version
Fedora 26 with latest updates (expat 2.2.3)

### Behavior
When starting the Appimage the following issue arises:

```
*** Error in `/tmp/.mount_Slic3r6nPQBC/usr/bin/bin/perl5.22.0': free(): invalid pointer: 0x0000000004dff9dc ***
```

The associated backtrace from gdb:
```
#0  0x00007fe3a66c366b in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007fe3a66c5470 in __GI_abort () at abort.c:89
#2  0x00007fe3a67098b1 in __libc_message (do_abort=do_abort@entry=2, fmt=fmt@entry=0x7fe3a6826100 "*** Error in `%s': %s: 0x%s ***\n")
    at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007fe3a6714759 in malloc_printerr (ar_ptr=<optimized out>, ptr=<optimized out>, str=0x7fe3a6822d6e "free(): invalid pointer", action=<optimized out>)
    at malloc.c:5077
#4  0x00007fe3a6714759 in _int_free (av=<optimized out>, p=<optimized out>, have_lock=0) at malloc.c:3873
#5  0x00007fe3a671a0be in __GI___libc_free (mem=<optimized out>)
    at malloc.c:2947
#6  0x00007fe3956861cc in XML_ParserFree (parser=0x361a6b0)
    at lib/xmlparse.c:1408
#7  0x00007fe38f115cec in driParseOptionInfo (info=info@entry=0x7fffffffbc10, configOptions=configOptions@entry=0x7fe38f123ca0 <__driConfigOptionsLoader> "<driinfo>\n<section>\n<description lang=\"en\" text=\"Initialization\"/>\n<description lang=\"ca\" text=\"Inicialització\"/>\n<description lang=\"de\" text=\"Initialization\"/>\n<description lang=\"es\" text=\"Inicializ"...)
    at xmlconfig.c:756
#8  0x00007fe38f1138d9 in loader_get_dri_config_device_id () at loader.c:100
#9  0x00007fe38f1138d9 in loader_get_user_preferred_fd (default_fd=5, different_device=different_device@entry=0x360341c) at loader.c:170
#10 0x00007fe38f1118fd in dri3_create_screen (screen=0, priv=0x3600e60)
    at dri3_glx.c:815
#11 0x00007fe38f0e5cc6 in AllocAndFetchScreenConfigs (priv=0x3600e60, dpy=0x1523400) at glxext.c:806
#12 0x00007fe38f0e5cc6 in __glXInitialize (dpy=dpy@entry=0x1523400)
    at glxext.c:932
#13 0x00007fe38f0e1a62 in glXGetFBConfigs (dpy=0x1523400, screen=0, nelements=nelements@entry=0x7fffffffbe5c) at glxcmds.c:1670
#14 0x00007fe38f0e2d22 in glXChooseFBConfig (dpy=<optimized out>, screen=<optimized out>, attribList=0x7fffffffbfb0, nitems=0x7fffffffbfac) at glxcmds.c:1608
#15 0x00007fe39c13e451 in glXChooseFBConfig (dpy=0x1523400, screen=0, attrib_list=0x7fffffffbfb0, nelements=0x7fffffffbfac) at libglx.c:1438
#16 0x00007fe39d693bd9 in wxGLCanvasX11::InitXVisualInfo(int const*, __GLXFBConfigRec***, XVisualInfo**) ()
    at /tmp/.mount_Slic3rUUNrim/usr/bin/bin/libwx_gtk2u_gl-3.0.so.0
#17 0x00007fe39d6925fb in wxGLCanvas::Create(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&, int const*, wxPalette const&) ()
    at /tmp/.mount_Slic3rUUNrim/usr/bin/bin/libwx_gtk2u_gl-3.0.so.0
#18 0x00007fe39d6927e3 in wxGLCanvas::wxGLCanvas(wxWindow*, int, int const*, wxPoint const&, wxSize const&, long, wxString const&, wxPalette const&) ()
    at /tmp/.mount_Slic3rUUNrim/usr/bin/bin/libwx_gtk2u_gl-3.0.so.0
#19 0x00007fe3904091d6 in XS_Wx__GLCanvas_newDefault(interpreter*, cv*) ()
    at /tmp/.mount_Slic3rUUNrim/usr/bin/lib/site_perl/5.22.0/auto/Wx/GLCanvas/GLCanvas.so
#20 0x00000000004bc77b in Perl_pp_entersub ()
#21 0x00000000004b54b6 in Perl_runops_standard ()
#22 0x000000000043c5b3 in Perl_call_sv ()
#23 0x00007fe390408758 in XS_Wx__GLCanvas_new(interpreter*, cv*) ()
    at /tmp/.mount_Slic3rUUNrim/usr/bin/lib/site_perl/5.22.0/auto/Wx/GLCanvas/GLCanvas.so
#24 0x00000000004bc77b in Perl_pp_entersub ()
#25 0x00000000004b54b6 in Perl_runops_standard ()
#26 0x000000000043c892 in Perl_call_sv ()
#27 0x00007fe39322a765 in XS_Wx___App_Start(interpreter*, cv*) ()
    at /tmp/.mount_Slic3rUUNrim/usr/bin/lib/site_perl/5.22.0/auto/Wx/Wx.so
#28 0x00000000004bc77b in Perl_pp_entersub ()
#29 0x00000000004b54b6 in Perl_runops_standard ()
#30 0x0000000000443d69 in perl_run ()
#31 0x000000000041d7bb in main ()
```

Downgrading to 2.2.1 from [koji](https://koji.fedoraproject.org/koji/buildinfo?buildID=915885) or to the base 2.2.0 via `dnf downgrade expat` will work around the issue. I suspect that building that Appimage with expat 2.2.2 or 2.2.3 might fix the issue and be backwards compatible with host os's that have a lower expat version.

jkoelker, 2017-08-13:
I should also point out that upstream's Appimage does not exhibit the same issue, so figuring out the delta between the two might give additional clues. I couldn't find the instructions/script for generating the Appimage that this fork uses, so I was unable to investigate that further.

thed0ct0r, 2017-08-13:
i can confirm the same problem with the same error message on arch linux with expat > v2.2.1.
same solution (downgrading to v2.2.1) works as well.

bubnikv, 2017-08-14:
There seems to be two issues here:

First, one of the bundled wxwidget libraries links against libexpat,
second, the Slic3r XS.so links against a static version of libexpat.

I suppose I will have to bundle the correct version of libexpact into the
appimage for now.

bubnikv, 2017-08-14:
It seems it crashes inside libexpat, which is called from the X11 opengl libraries. Frankly I am not sure how to fix that. As libexpat is called by the local X11 libraries and I cannot just bundle the complete Linux into the appimage, libexpat is expected to have a stable binary API and not to be bundled with the AppImage.

Currently Slic3r PE bundles and links expat statically even on Linux, and I can change that. But that should not make any difference as the bundled libexpat is only used for AMF format import / export.

> I should also point out that upstream's Appimage does not exhibit the same issue, so figuring out the delta between the two might give additional clues. I couldn't find the instructions/script for generating the Appimage that this fork uses, so I was unable to investigate that further.

You can check the content of the two appimages, there may be the differences in the set of bundled libs. Also there will be the subtle differences (read the Linux shared library hell) in the versions of the bundled libraries.

bubnikv, 2017-08-29:
Would you try this build, please? I have changed the libexpat linkage to dynamic and I am letting Slic3r use the system wide libexpat. Frankly I am not sure whether this is the right solution. The system wide libexpat seems to be loaded by some of the X11 modules, so it seems to be a good idea to not load any other libexpat into the same process. @probonopd ?
 
[Slic3r-1.37.0.36-prusa3d-linux64-full-g4bd694e-201708291445.AppImage.zip](https://github.com/prusa3d/Slic3r/files/1261554/Slic3r-1.37.0.36-prusa3d-linux64-full-g4bd694e-201708291445.AppImage.zip)

awigen, 2017-08-29:
Fixes the issue for me!

probonopd, 2017-08-29:
Correct @bubnikv, libexpat shouldn __not__be bundled inside the AppImage, like all libraries listed in the [excludelist](https://github.com/AppImage/AppImages/blob/master/excludelist). Instead, these libraries should be assumed to be present in the target systems and be loaded from there.

probonopd, 2017-08-29:
> Would you try this build, please?

Runs on openSUSE LEAP 42.3.

bubnikv, 2017-08-29:
@probonopd thanks for the hint. I see the bundling is a pretty fuzzy topic. Now after you introduced the AppImage, you ought to talk the distribution maintainers into maintaining the binary compatibility :-)

Closing.

closed 2017-08-29 by bubnikv