prusa3d/PrusaSlicer #473: Adaptive Layer Height not respecting Mininimum layer height limit resolution

theopensourcerer, 2017-08-14:
### Version
1.36.0-prusa3d-linux64
### Operating system type + version
Ubuntu 16.04 LTS 64bit

### Behavior
The new Adaptive Slicing is generating gcode which does not respect the Min: parameter in the Layer height limits area of the Printer Settings tab.

My min height is set to 0.04. The gcode had my printer trying to print at all sorts of odd heights including, for example: 31.33

The very fine increments appeared to be at 0.05 :-(

[image]
[image]

Hugoz12, 2017-08-15:
Can you upload the stl file + config and maybe the gcode?
Do you have support activated? (Something similar happened to me when having support activated as.)

bubnikv, 2017-08-15:
The support layers are stepped independently of the object layers.

theopensourcerer, 2017-08-15:
@Hugoz12 Thanks and sure thing. Config bundle, STL and generated gcode attached. I didn't use supports.

[Slic3r_config_bundle.ini.zip](https://github.com/prusa3d/Slic3r/files/1224417/Slic3r_config_bundle.ini.zip)
[bearing-mount.stl.zip](https://github.com/prusa3d/Slic3r/files/1224410/bearing-mount.stl.zip)
[bearing-mount.gcode.zip](https://github.com/prusa3d/Slic3r/files/1224411/bearing-mount.gcode.zip)

DrLex0, 2017-09-11:
The current tooltip for the Min field seems to imply that filling in a value of 0.04 would also cause all Z coordinates to be rounded to 0.04mm precision. Slic3r definitely does not honor this, because if I quickly try it with Min set to “0.1”, I still find commands like “G1 Z1.237” in the output. The heights do increase with no less than 0.1mm, but there is some offset that is not rounded to that precision. Even if the printer will round the Z coordinates, then still the extrusion rates would have been calculated for the non-rounded value.

I think we also need more configurability here: resolution should be configurable independently of minimum layer height, therefore I created #513. If that would be properly implemented, this issue will automatically be fixed too.

bubnikv, 2017-10-03:
> The current tooltip for the Min field seems to imply that filling in a value of 0.04 would also cause all Z coordinates to be rounded to 0.04mm precision.

If this is how the label is understood, I should re-fraze it. The minimum layer height is exactly what it says: Thinner layers will not be extruded.

@theopensourcerer Does the current behavior cause any issues for you? I admit we test Slic3r on our machines with 16x micro stepping on Z axis, so the fine layer stepping is not an issue to us.

theopensourcerer, 2017-10-03:
Thanks for the follow up.

To answer your question, yes, I think it does actually. When my printer tries to move to Z heights that are it cannot accurately do, it seems to get rather messed up. I get more rough surfaces and sometimes it gets so bad that I end up with blobs of plastic which then cause the nozzle to "bounce" or the piece gets knocked when going to the next layer.

Maybe I just have a rubbish printer ;-) (Anet A8) But when I use precise heights in Slic3r I can get very nice prints in both PLA and PETG.

bubnikv, 2017-10-03:
I think what you likely want is https://github.com/prusa3d/Slic3r/issues/202

Do you know what stepping is allowed on your Z axis?

theopensourcerer, 2017-10-03:
Yes, it's 0.04mm. That's why I entered it in the box as I thought that was what the field was for ;-)

#202 sounds exactly right :-) 

So my issue is about field labelling and descriptions, but I think #202 is the root of the problem.

theopensourcerer, 2017-10-03:
It's 400 steps per mm according to M503:
`Recv: EPR:3 11 400.0000 Z-axis steps per mm`

bubnikv, 2017-10-03:
> It's 400 steps per mm according to M503:

That is exactly the same number we have with our Prusa i3 MK2, and we have no issue with the Z axis resolution.

It may be, that your machine does not follow the micro steps accurately, meaning that the micro steps are highly non linear. I guess your machine is configured with 16x micro steps, so a full step is 0.04mm, and the full step distance is repeatable.

You may try to increase the Z motor current.

Anyway, I am closing this issue, and I will keep https://github.com/prusa3d/Slic3r/issues/202 open.
I will reference this issue from https://github.com/prusa3d/Slic3r/issues/202

closed 2017-10-03 by bubnikv