prusa3d/PrusaSlicer #474: Slic3r will not get past go

cognitdiss, 2017-08-14:
### Version
_Version of Slic3r used goes here_
1.36.2-prusa3d-stable-win64, installed via fresh DL from Prusa website

_Use `About->About Slic3r` for release versions_
NA

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
NA

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 Pro via Bootcamp 2016 MBP

### Behavior
* _Describe the problem_
Slic3r imports stl or obj successfully, then hangs on either after slice or gcode button press, windows then quits it out.
--note, the file attached is larger than print bed, failure also occurs after 80% scale.
* _Steps needed to reproduce the problem_
press slice or gcode button
  * _If this is a command-line slicing issue, include the options used_
NA
* _Expected Results_
anything
* _Actual Results_
nothing
  * _Screenshots from __*Slic3r*__ preview are preferred_
Done

[image]

_Is this a new feature request?_
no
#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
Done

[SquareSlicer.zip](https://github.com/prusa3d/Slic3r/files/1223248/SquareSlicer.zip)

[config.zip](https://github.com/prusa3d/Slic3r/files/1223253/config.zip)

Hugoz12, 2017-08-15:
The STL seems to have inch instead of mm as a unit. If you look at your screenshot it says 9.59x9.59x0.1 (unit always mm).

 Scale it by 2540% and it works fine for me.

The crash seems to be caused by the height. If you set your layers height to a lower value than the object it works even with the file you provided. (I.e. 0.05mm)

bubnikv, 2017-08-15:
> Scale it by 2540% and it works fine for me

Still it should not crash hard. I will look into it.

Hugoz12, 2017-08-15:
Yep it shouldn't. Was just analyizing what could have the casue for it and a quick work around. ;)

closed 2017-08-15 by cognitdiss