prusa3d/PrusaSlicer #475: Different fill patterns for top/bottom

m-Th, 2017-08-15:
### Version
Slic3r Prusa Edition
Version: 1.36.2-prusa3d-stable-win64
Build: Slic3r-1.36.2-prusa3d-stable-win64-full-201707281104
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 6.1.7601] 

### Behavior
Now we have in Print Settings -> Infill a single option for _Top/Bottom Infill pattern_.

Please make two separate ones for bottom and top. 

This feature has various benefits especially for functional prints. For example it will allow us to speedup the print on the part which isn't exposed to the user while the other would have a certain finish.

Referenced also here: https://github.com/prusa3d/Slic3r/issues/458#issuecomment-322199111

_Is this a new feature request?_
Yes.

konvoj, 2017-09-06:
Please also consider the separate extrusion width only for the top layer to allow this setting:
https://ultimaker.com/en/community/50643-neosanding-20-faster-cleaner-top-layers-but-different

hbogs, 2018-01-21:
Looks like different fill pattern for top/bottom has been implemented upstream (https://github.com/alexrj/Slic3r/issues/945) - any possibility of getting it merged to this fork?

supermerill, 2018-01-24:
As I also need this, i made this split, with a new code.
[image]

Maybe taking the diff from alexrj#945 be better.
My code: https://github.com/supermerill/Slic3r/tree/top_bottom_split_infill
Should i make a push request or someone is up to the task to pick the commit from alexjr branch?

hbogs, 2018-03-19:
@supermerill Nice :) Maybe open up a PR and see if its possible to get it merged upstream?

bubnikv, 2018-03-20:
We will discuss this feature internally first. The general rule is to try
to keep the number of parameters low, so we need to justify adding another
parameter.

bubnikv, 2019-02-22:
Implemented with ab30370fb4bfa92d8ae77419686e2b6cde923fcf
Thanks @supermerill for the initial implementation.
Closing.

closed 2019-02-22 by bubnikv